flipperdevices/flipper-hackathon-moscow #10: 2048 Game

OlegSchwann, 2021-11-18:
### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Простая игра [2048](https://2048game.com/) подходит под набор кнопок и разрешение экрана Флиппера. [Достаточно популярна.](https://en.wikipedia.org/wiki/2048_(video_game)). Есть [открытые реализации на С](https://github.com/elgin9507/2048) для прототипа.

### Готовый проект будет уметь...

На поле, попиксельно отрисованном как тут:
[image]
происходит игра как тут:
https://2048game.com/
Есть базовые анимации сдвига плиток, 3 промежуточных кадра.
При проигрыше можно выйти или начать заново, состояние не сохраняется.

hedger, 2021-11-18:
А порт дума?

OlegSchwann, 2021-11-18:
> А порт дума?
[r/ItRunsDoom](https://www.reddit.com/r/itrunsdoom/top/?t=all) intensifies.
Только если через [хак с трансляцией экрана с более мощного устройства](https://twitter.com/Foone/status/1302287398949142533) в целях саморекламы.

hedger, 2021-11-18:
STM32F429 тянет дум в цвете с неплохим разрешением, тут же хоть частота контроллера в 3 раза ниже, но и экран попроще. Должно смочь

zhovner, 2021-11-18:
Мне нравится.

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

OlegSchwann, 2021-11-30:
Я зарегистрировался, ждем @Sefjor .

Sefjor, 2021-12-01:
зарегился