flipperdevices/flipper-hackathon-moscow #11: Фаззинг управляющего ПО

LuciyVI, 2021-11-18:
### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Фаззинг сможет выявлять баги и потенциальные уязвимости управляющего ПО.

### Готовый проект будет уметь...

При сборке прошивки будет будет добавляться инструментация.на ПО будут подоваться случайно сгенерированные данные

OlegSchwann, 2021-11-18:
> ```$ /usr/bin/time --format='%E\n' make all > /dev/null;```
> ```1:37```
При прохождении линтеров пожалуйста, сборка и так относительно долгая.

LuciyVI, 2021-11-18:
Сборка для фаззинга не являтся продуктовой версией

aprosvetova, 2021-11-19:
Не можем понять, о чем проект, и какова его цель.

closed 2021-11-19 by aprosvetova