flipperdevices/flipper-hackathon-moscow #13: Радио мессенджер (Sub-GHz)

lomalkin, 2021-11-18:
### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

<s>Уже джва года</s> Хочу сделать чат-мессенджер, позволяющий передавать простые текстовые сообщения между устройствами.
Т.к. на самом устройстве нет полноценной клавиатуры - используем для этого VCP-CLI интерфейс и мобильные клиенты самого Флиппера.

Сценарий:
Можно сидеть в офисе и переписываться по радио, без интернета, смс и регистрации, открыв консоль Флиппера с компа или в походе на природе, через приложение-клиент (подключенный к Флипперу по BT) с мобильника, пока сам Флиппер лежит в рюкзаке.

Фичи:

1. Прием сообщений в фоне
2. Уведомление о новом сообщении по вибрации (после чего уже можно открыть его с телефона или компа)
3. Вместо имени собеседника используется уникальное имя самого Флиппера
4. Простая проверка аутентичности сообщений через подпись крипто-ключом (MAC).
5. Сохранение чат-лога (всех принятых и переданных сообщений) на SD карту, опционально. Добавление в лог таймстампа по RTC.
6. <s>Костыльная</s> реализация переповтора отправки сообщения в течении некоторого времени после отправки. Нумерация каждого сообщения уникальным id для этого. (с этим вообще весело будет)

Используется встроенный модуль SubGHz (выбрать разрешенную частоту, но позволяющую реализовать наиболее дальнобойную связь в условиях города). Хорошо бы иметь возможность использования модуля в режиме приема в 

Это просто обязано быть в основной прошивке =)

### Готовый проект будет уметь...

Вероятно, я cмогу успеть только сделать некий PoC - а именно:
- Команду для ожидания и приема сообщений через CLI
- Возможность отправки сообщений через тот же CLI по радио от имени Флиппера

Не будет точно:
- Фонового режима (для приема сообщений без предварительно запущенной команды)
- Работы с SD картой, переповторами
- Поддержки в мобильных клиентах

P.S. Если у идеи найдутся единомышленники - велкам в комменты. Очень хочу такую штуку.

aprosvetova, 2021-11-18:
Супер!

karasevIA, 2021-11-19:
Проект действительно интересен, я бы вписался

Sefjor, 2021-11-19:
Хотелось бы поучаствовать, смогу взять любую часть этого функционала. На Github не смотрите, с тех пор 6 лет пишу коммерческий софт на embedded C, C++, Qt5

DakotaMitch, 2021-11-19:
А в mesh-сеть соедините всех флипперов, чтобы покрыть связью весь город? 
Такое сделали в мешастике, но у них мало устройств https://github.com/meshtastic

alexanderkulagin, 2021-11-19:
Отличная идея

tonyfreeman, 2021-11-19:
Готов взять мобильную часть

karasevIA, 2021-11-19:
Хорошая идея писать что можем сделать
Я с радио хорошо знаком, могу эту часть на себя взять

lomalkin, 2021-11-29:
Данная заявка снимается, т.к. уже существует подобная реализация в CLI.

closed 2021-11-29 by lomalkin

lomalkin, 2021-11-29:
https://github.com/flipperdevices/flipperzero-firmware/pull/851