flipperdevices/flipper-hackathon-moscow #14: Реализация Openhaystack для трекинга устройства через сеть Apple Find My

kinjalik, 2021-11-18:
### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Данный проект решает задачу трекинга Flipper на местности с использованием сети Apple Find My. Для достижения желаемого результата используется протокол [Openhaystack](https://github.com/seemoo-lab/openhaystack)  на основе Bluetooth LE. Он использует окружающие Apple-устройства ( iPhone, iPad, etc ) для определения примерного местоположения.

### Готовый проект будет уметь...

1. Запускаться на Flipper
2. Принимать ключ шифрования пакетов для Apple My на устройство
3. Отображать примерное местоположение в интерфейсе эталонного клиента Openhaystack

aprosvetova, 2021-11-18:
Главным приключением в этом проекте будет необходимость сформировать корректный advertisement data пакет так, чтобы при этом не нарушалась штатная функциональность BLE. Если у вас получится, будет волшебство. 

Я заводила PoC такого трекинга на Флиппере, он отображался в клиенте Openhaystack, так что тегайте в Телеграм-чате, если будут вопросы по структуре пакета. Но поселить эту фичу рядом с нашим штатным блютусом будет трудно.

kinjalik, 2021-11-20:
Укомплектовали команду :)

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

AlexStrNik, 2021-11-30:
Зарегистрировался

genom10, 2021-12-01:
Зарегистрировался