flipperdevices/flipper-hackathon-moscow #15: Таймер для Флиппера

Astrrra, 2021-11-18:
### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Буквально таймер, как на телефоне. Полезен в ситуациях когда телефон не под рукой или занят (как раз часто бывает что на телефоне нельзя поставить таймер потому что на нем будет происходить что-то для чего таймер и нужен, а в фоне его поставить не вариант)

### Готовый проект будет уметь...

Выбор времени от 1с до 23:59:59
Выбор видов сигнала (звук и/или вибрация)
Обратный отсчёт
Сигнал при истечении времени
Отключение сигнала

### GitHub второго участника

@Vital7

### GitHub третьего участника

@SunnyCapt

Astrrra, 2021-11-18:
Если хватит времени то еще должно быть довольно просто добавить секундомер, но это уже немного не уверена успеем ли за 3 дня с новым железом

aprosvetova, 2021-11-19:
Решили, что слабовато для троих(
Придумайте что-нибудь посмелее!

Astrrra, 2021-11-19:
Ну раз посмелее...

LionZXY, 2021-11-19:
Можете присоединиться к идеи сделать часы на флиппере

closed 2021-11-19 by aprosvetova