flipperdevices/flipper-hackathon-moscow #2: Вибромассажер с удаленным управлением по Bluetooth

zhovner, 2021-11-18:
### Вы капитан команды?

- [X] Yes, I am a leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Так как во Флиппере есть вибромотор, его можно использовать как массажер скрытого ношения для интимных мест. Я хочу сделать приложение для мобилы, в котором можно будет удаленно управлять вибромоторчиком на Флиппере. То есть двигать ползунок на телефоне и менять силу вибрации.

### Готовый проект будет уметь...

Приложение на мобильном телефоне, которое подключается к флипперу и включает вибромотор. Сила вибрации регулируется на телефоне ползунком от нуля до максимума.

aprosvetova, 2021-11-18:
Павел, отличная идея.
Скажите, вам в команду нужны тестировщики?

Astrrra, 2021-11-18:
Эх, а идея-то с потенциалом изменить мир была...

LionZXY, 2021-11-18:
Кажется, мощность BLE сигнала будет низкая при ношении. Мне кажется вам потребуется в команду hardware разработчик для дополнительной платы, которая будет усиливать сигнал

NetBUG, 2021-11-18:
@aprosvetova можно ли создать issue на доработку hardware в следующей ревизии Флиппера?

lomalkin, 2021-11-18:
@LionZXY можно проводной сделать и подключать к телефону через OTG / typeC to typeC.

Astrrra, 2021-11-18:
> @LionZXY можно проводной сделать и подключать к телефону через OTG / typeC to typeC.

Есть оч частый юзкейс когда контроль у партнёра, там провод далеко не всегда подходит

LionZXY, 2021-11-18:
@lomalkin я уже решил проблему, засунув в себя телефон. Теперь ищу способ управлять удаленно телефоном

NetBUG, 2021-11-18:
Говори номер

bytesex, 2021-11-19:
Помогу решить проблему стимуляции интимных зон без этих ваших Флипперов.
Засунувшим в себя телефон респект за тренированность слотов

closed 2021-11-19 by aprosvetova

NetBUG, 2021-11-19:
Жаль, жаль :(

NetBUG, 2021-11-24:
[image]