flipperdevices/flipper-hackathon-moscow #3: Менеджер SPI флешек

hedger, 2021-11-18:
### Вы капитан команды?

- [X] Yes, I am a leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Хотел бы сделать модуль для флиппера, суть такова...  

Пользователь может читать и шить SPI-флешки через прищепку или запаяв на макетку, вставляющуюся в гребёнку.
Опционально выполнять верификацию записанного, сравнивая с оригинальным файлом. 
Также сделать отображение параметров подключённой флешки - размер, вендор, ...
Поддерживаются флешки с уровнями 3.3 вольта. 

В составе модуля на флиппере будет простой менеджер файлов дампов на SD-карте.

Если с реализацией базового функционала не будет затыков, попробуем добавить связку с приложением под андроид для управления образами.

Модуль после доработки до приличного состояния можно будет включить в стоковую прошивку, т.к. функционал востребованный.

P.S. Второй год жду флиппер

### Готовый проект будет уметь...

Читать SPI флешки, сохранять образ на SD
Писать SPI флешки с образа на SD
Определять параметры подключённой флешки

### GitHub второго участника

@ghettorce

zhovner, 2021-11-18:
Предполагается ли GUI на самом экранчике флиппера для этой программы?

hedger, 2021-11-18:
Да, для отображения параметров флешки, выбора образа, вывода прогресса операции.

aprosvetova, 2021-11-18:
Пожалуйста, добавьте про GUI для управления всеми этими фичами в раздел про критерии законченности.
Идея отличная!

LionZXY, 2021-11-18:
Очень крутая идея!

zhovner, 2021-11-18:
Хочется такую штуку для сброса пароля UEFI на макбуках https://www.youtube.com/watch?v=DrRzyLFdj9o

hedger, 2021-11-18:
> Хочется такую штуку для сброса пароля UEFI на макбуках https://www.youtube.com/watch?v=DrRzyLFdj9o

На видео мелькает SCL/SDA - это ж I2C, но если будет рабочий каркас UI с функционалом под SPI, можно будет его тоже вкрутить

x893, 2021-11-18:
Как можно присоединиться ?

hedger, 2021-11-18:
> Как можно присоединиться ?
Написал в телегу

hedger, 2021-11-18:
Апдейт: в команде +1, возможно, вытянем I2C в рамках хакатона. Возможности прикидываем

x893, 2021-11-22:
Была информаця, что можно прислать firmware.bin для проверки на флиппере. Я сделал текущий dev (NO_BOOTLOADER) под Segger Embedded Studio (windows) и хотел проверить перед внесением дополнений. Можно загрузить прошвку и сказать - работает или нет ?

zhovner, 2021-11-22:
@x893 спросите в чате пожалуйта. Сразу в одном сообщинии приложите прошивку и напишите что делать для проверки.

alexey-gamov, 2021-11-25:
В качестве реферанса можно глянуть **XHorse chip programmer**

hedger, 2021-11-30:
Зарегистрировался на Timepad

ghettorce, 2021-11-30:
Зарегистрировался на Timepad

x893, 2021-12-02:
Регистрируюсь