flipperdevices/flipper-hackathon-moscow #4: "Змейка"

terge-eg, 2021-11-18:
### Вы капитан команды?

- [X] Yes, I am a leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Классическая "Змейка"
https://en.wikipedia.org/wiki/Snake_(video_game_genre)
Используются только кнопки и дисплей

### Готовый проект будет уметь...

 В игре будет 10 уровней, аналогичных классической версии для Nokia 3310 + генератор уровней.

LionZXY, 2021-11-18:
https://github.com/flipperdevices/flipperzero-firmware/pull/829

zhovner, 2021-11-18:
Змейка уже есть https://github.com/flipperdevices/flipperzero-firmware/pull/829

closed 2021-11-19 by aprosvetova