flipperdevices/flipper-hackathon-moscow #5: Ground Control Station для квадрокоптера

okalachev, 2021-11-18:
### Вы капитан команды?

- [X] Yes, I am a leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Подсоединение к квадрокоптеру через Wi-Fi по протоколу MAVLink. Отображение телеметрии. Управление квадрокоптером стиком флиппера.

### Готовый проект будет уметь...

Показывать телеметрию, управлять дроном, еще какие-нибудь приколы.

aprosvetova, 2021-11-18:
Во Флиппере нет поддержки Wi-Fi. Вы планируете реализовать это через наш внешний модуль ESP32-S2?

okalachev, 2021-11-18:
Упс, обидно. Можно и внешний модуль, хотя не так прикольно. Есть еще телеметрия на 433 МГц, но там все намного сложнее...

zhovner, 2021-11-18:
Как я понял, вы говорили только по 2,4GHz. Или на 433 тоже актуально?

okalachev, 2021-11-18:
Как вариант, собственно, связь через внешний ESP. Есть телеметрийный радиомодуль на 433 МГц, но вот сможет ли Флиппер вот так взять, и к нему подключиться - сложно сказать.

Skorpionm, 2021-11-18:
Если дрона не жалко можно делать через wifi
Пульты для дронов хоть и используют 2,4g но протокол там не тср. А полу сырой и ему не страшны потери пакетов.

okalachev, 2021-11-18:
А я не про пульт, а про телеметрию. А она может работать очень много как, хоть по 4G.

aprosvetova, 2021-11-19:
Пока это абстрактная и очень непонятная идея :(
Попробуйте придумать что-нибудь еще.

closed 2021-11-19 by aprosvetova