flipperdevices/flipper-hackathon-moscow #6: Часы на главный экран

OlegSchwann, 2021-11-18:
### Вы капитан команды?

- [X] Yes, I am a leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Привык, что время отображается на любом экране любого устройства. Ноутбук, телефон, книга, VR очки, микроволновка. Только Флиппер не показывает время, несколько раз брал его, что бы посмотреть только на неизменного дельфина.

Хочу добавить отображение времени на главный экран с дельфином.

### Готовый проект будет уметь...

В настройках можно установить время.
На главном экране оно отображается в 24 часовом формате.
При перезагрузке ( ◀️ + ↩️ ) и отключении время не сбрасывается.

OlegSchwann, 2021-11-18:
Подскажите, где можно посмотреть схему устройства в поисках RTC?

skotopes, 2021-11-18:
RTC встроен в MCU, посмотреть можно в его референс мануале.

Уже есть:

- Питание в обход чарджера (домен питается пока батарея не уйдет в защиту)
- Инициализация RTC домене при старте
- Комманда в консоле для просмотра и установки времени

Программа минимум: 

- FuriHalRtc, по аналогии с другими подсистемами нужно имплментировать и утащить туда инициализацию из rtc.c
- Gui/Settings: настройки часов и модули для работы с ними
- Rpc: получение и установка времени

Опционально:

- UI/UX: статус бар слишком маленький и для часов место не планировалось ни в каком виде вообще. Нужно понять как и куда их вписывать и как это будет выглядеть
- Сохранение времени при ресете RTC домена

aprosvetova, 2021-11-19:
Проект необычный. Интерфейс получается простой, а основная работа относится к FuriHal и внутренним API.
Мы в любом случае планируем FuriHalRtc и решили, что будет лучше, если его заведёт наша команда аналогично с остальными подсистемами, так как это компонент ядра.

closed 2021-11-19 by aprosvetova