flipperdevices/flipper-hackathon-moscow #7: Транслятор сигналов ИК-пульта в нажатия клавиш USB HID

ashpooljh, 2021-11-18:
### Вы капитан команды?

- [X] Yes, I am a leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Представим ситуацию: вы собрались посмотреть сериал на своём ноутбуке, который вы используете как импровизированный домашний кинотеатр. Подключив всю необходимую периферию (телевизор, USB-аудиокарту, зарядное устройство...) и запустив воспроизведение серии в каком-нибудь [VLC](https://www.videolan.org/vlc/), вы оставили ваш ПК на тумбочке, а сами расположились на диване в другом конце комнаты. 

**Внимание, вопрос.** Как управлять воспроизведением (пауза, перемотка, переключение серий...) без необходимости:
1) вставать с дивана,
2) подходить к ноутбуку,
3) открывать крышку,
4) трогать клавиатуру пальцами в крошках от чипсов,
5) перетаскивать окно проигрывателя между мониторами?

**Решение.** Научить Флиппер распознавать нажатия клавиш на пульте от телевизора (научили уже 🤔) и, прикинувшись USB-клавиатурой, преобразовывать их в сочетания клавиш, выполняющие соответствующие действия в вашем видеопроигрывателе. Модуль должен поддерживать разные пульты и переводить их в сочетания клавиш разных популярных программ.

**Проблемы.** Если в качестве большого экрана используется телевизор, скорее всего, придётся использовать другой пульт либо заклеивать ИК-приёмник телевизора, иначе он тоже будет реагировать на сигналы пульта. Потом можно добавить поддержку LPD433-пультов, но это за рамками хакатона.

**Используемые возможности Флиппера.** Очевидно, ИК-приёмопередатчик и режим BadUSB. Подгрузка таблиц преобразований будет осуществляться из файлов на SD-карточке.

**Претензия на включение в официальную прошивку.** Да. На самом деле, я удивлюсь если там этого уже нет, лень было проверять.

### Готовый проект будет уметь...

Готовый модуль должен для нескольких пар "пульт-программа" продемонстрировать в реальном времени приём сигналов пульта и отправку соответствующих комбинаций клавиш на ПК согласно таблице преобразований. Должны быть реализованы таблицы как минимум для двух пультов и двух популярных программ-видеопроигрывателей для ПК, а также их загрузка с SD-карточки и выбор нужной комбинации с помощью GUI Флиппера.

### GitHub второго участника

@Elecon-rou

zhovner, 2021-11-18:
Очень крутая идея, мы до такого не додумались.

aprosvetova, 2021-11-18:
Идея супер! Но у нас, кажется, сейчас нет удобных API для работы с BadUSB вне штатного приложения. Могу ошибаться. @nminaylov?

ashpooljh, 2021-11-18:
Ну, вот и API-шку накидаем заодно :)

nminaylov, 2021-11-18:
Есть API HID мыши и клавиатуры, можно слать нажатия кнопок и движения мыши. Единственное что клавиатура пока стандартная, на 101 клавишу. Чтобы управлять плеером надо немного поправить дескриптор

Elecon-rou, 2021-11-30:
Зарегистрировался

trexxet, 2021-11-30:
Зарегистрировался

ashpooljh, 2021-12-05:
Реализация: https://github.com/rwxecut/flipperzero-firmware-irusb/tree/irusb