flipperdevices/flipper-hackathon-moscow #9: Отображение любых данных с сервера

sapoarty, 2021-11-18:
### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Хотелось бы реализовать отображение на экране любых данных с заданного сервера по API. Аналог настольных часов LaMetric Time. При подключении к сети в любое время можно увидеть необходимую информацию, например:
- Температура
- Время
- Курс валют
- Количество посетителей сайта
- Новости
- Все, что угодно
При необходимости, устройство можно оставить подключены к питанию.

### Готовый проект будет уметь...

Отображать значение заданного параметра с заданного пользователем сервера на главном экране в режиме ожидания. При превышении или занижении значения, приходит уведомление в виде вибрации и/или мигания экрана.

LionZXY, 2021-11-18:
Во флиппере нет возможности использовать сеть, так как нет WiFi

OlegSchwann, 2021-11-18:
Критика, основанная на опыте использования умных часов:
проще достать телефон, чем что-то читать на таком мелком экране.
На Флиппере всего 4 строчки помещается, 21 символ в ширину, стимул просто достать телефон еще яснее.

skotopes, 2021-11-18:
В Rpc есть возможность стримить фрэйм буфер на флиппер. На самом всё уже есть.

sapoarty, 2021-11-18:
> Критика, основанная на опыте использования умных часов:
> проще достать телефон, чем что-то читать на таком мелком экране.
> На Флиппере всего 4 строчки помещается, 21 символ в ширину, стимул просто достать телефон еще яснее.

Да, в случае умных часов есть такое неудобство. Однако, в данном случае, можно отображать данные одной строкой во всю высоту экрана(для большинства данных достаточно), и, при необходимости, использовать бегущую строку. Также нет необходимости "пробуждать" устройство, или производить какие-либо действия для отображения, дисплей всегда перед глазами, на столе.

aprosvetova, 2021-11-19:
Как подметил @ skotopes, на Флиппер уже можно стримить изображения, используя его как виртуальный дисплей.
[Пример работы функции.](https://www.youtube.com/watch?v=u016RvMcA0U)
В [go-flipper](https://github.com/flipperdevices/go-flipper), библиотеке для работы с RPC, эта функция реализована.
Как будет время, подготовлю утилиту, чтобы любой мог это протестить.

closed 2021-11-19 by aprosvetova