flipperdevices/flipper-hackathon-moscow #16: Тетрис

alexgr13, 2021-11-19:
### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Игра [Тетрис](https://ru.wikipedia.org/wiki/%D0%A2%D0%B5%D1%82%D1%80%D0%B8%D1%81)
Игровое поле размером 10 на 20 ячеек, ячейка размером 3*3 пикселя.
Сбоку выводится текущая скорость, количество очков и следующая фигура.

### Готовый проект будет уметь...

Есть скорости от 1 до 9, можно выбирать стартовую скорость перед началом игры.
Стрелками можно управлять падающей фигурой.
Линии исчезают и очки начисляются по правилам тетриса (1 линия — 1 очка, 2 линии — 3 очка, 3 линии — 7 очков, 4 линии — 15 очков).
Каждые 100 очков скорость увеличивается на единицу, максимум - 9.

### GitHub второго участника

@b1rd

LionZXY, 2021-11-19:
FYI: тетрис очень бдит за соблюдением лицензионных прав, скорее всего, в прошивку смерджить не сможем

aprosvetova, 2021-11-19:
Закрыто в пользу #17.

closed 2021-11-19 by aprosvetova