flipperdevices/flipper-hackathon-moscow #17: Танчики (Battle city)

alexgr13, 2021-11-19:
### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Упрощённая версия игры [Battle city](https://ru.wikipedia.org/wiki/Battle_City)
Игровое поле размером 12 на 12, на каждая клетка занимает 5*5 пикселей на экране.
В левой части экрана отображается игровое поле, в правой - информация (количество очков, количество оставшихся танков противника, количество оставшихся жизней).

### Готовый проект будет уметь...

Упрощения по сравнению с оригинальной игрой:

- Нет бонусов-улучшений
- Только один тип танков противника
- Нет передвижения на полклетки
- Нет типов клетки "кусты" и "лёд"
- Игровое поле уменьшено с 13*13 до 12*12

Необязательное, но хотелось бы успеть:

- Совместная игра вдвоём по Bluetooth

### GitHub второго участника

@b1rd

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

alexgr13, 2021-11-30:
Я зарегался

b1rd, 2021-11-30:
Я тоже 🐝

Kolobamanacas, 2021-11-30:
Я зарегался. (с)