flipperdevices/flipper-hackathon-moscow #19: Флиппер и твердотельный гироскоп

Avtopolet, 2021-11-19:
### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Хочу подключить к  Флипперу гироскоп типа MPU6050, написать приложение которое будет получать сырые данные с гироскопа и преобразоввывать их в углы(yaw, pitch, roll). 

### Готовый проект будет уметь...

Оценивая свои силы реально, для меня, критерием успеха будет вывести значения углов ориентации Флиппера в пространстве в градусах на его милый экранчик. Вторая цель-передача этих углов на другое устройство (ноутбук) и управление ориентацией объекта в ..скажем Blender.  Вращая Флиппером вращать модель.

factoryhero, 2021-11-19:
Это интересная идея. Если на внешний модуль добавить пару кнопок, то Флиппер можно превратить в Air Mouse или даже в 3DoF геймпад, как в Nintendo Switch.

alexanderkulagin, 2021-11-19:
В прошивке есть зачатки BLE HID, было бы очень круто реализовать Air Mouse. Но модуль с IMU нужно подготовить заранее — на месте не будет оборудования.

Avtopolet, 2021-11-19:
Да, конечно, платку с модулем подготовлю.

aprosvetova, 2021-11-19:
Подготовьте, пожалуйста, принципиальную схему для модуля, присоединяемого к гребёнке. Тогда одобрим проект и заранее пришлём вам протоборды для удобного прототипирования, чтобы выглядело хорошо и вставлялось удобно.

Avtopolet, 2021-11-19:
Модуль подключается напрямую, ток потребления модуля 500 мкА, все емкости на модуле не более одной мкФ. Замеры между VCC и GND показали не более 200 пФ, так что индуктивность не нужна.
[image]
[image]
[image]

DrZlo13, 2021-11-19:
Выглядит работоспособно, если на модуле есть подтяжки

lomalkin, 2021-11-19:
У вас модуль питается от 3.3V выхода Флиппера, но согласно нагугленной мною схеме самого модуля гироскопа - там есть LDO, а значит, что предполагается его работа от повышенного напряжения.
[image]
Варианты:
1. Продолжать использовать схему как у вас, но при сборке "соплей" закоротить вход и выход линейника на плате самого модуля, оставив конденсаторы и все остальное как есть.
2. Запитать модуль гироскопа от 5V выхода Флиппера. Логика его работы следующая: на нем всегда присутствует 5V, если питание приходит на Флиппер по USB, либо этот выход можно принудительно программно включить при работе от батареи из прошивки или графического интерфейса.
[image]

aprosvetova, 2021-11-20:
❗️При разработке внешнего модуля вам могут сильно помочь эти инструкции и чертежи. Обратите внимание, там есть ссылки на исходники шаблона модуля в EasyEDA и файлы контуров плат. 

https://docs.flipperzero.one/ru/development/external-module/guide

https://docs.flipperzero.one/ru/development/ext-modules

Avtopolet, 2021-11-20:
Именно этими шаблонами я и воспользовался

Avtopolet, 2021-11-20:
> ❗️При разработке внешнего модуля вам могут сильно помочь эти инструкции и чертежи. Обратите внимание, там есть ссылки на исходники шаблона модуля в EasyEDA и файлы контуров плат. [...]

Именно этими шаблонами я и воспользовался

Avtopolet, 2021-11-20:
> У вас модуль питается от 3.3V выхода Флиппера, но согласно нагугленной мною схеме самого модуля гироскопа - там есть LDO, а значит, что предполагается его работа от повышенного напряжения. [image] Варианты:
> 
> 1. Продолжать использовать схему как у вас, но при сборке "соплей" закоротить вход и выход линейника на плате самого модуля, оставив конденсаторы и все остальное как есть. (я бы выбрал этот вариант)
> 2. Запитать модуль гироскопа от 5V выхода Флиппера. Логика его работы следующая: на нем всегда присутствует 5V, если питание приходит на Флиппер по USB, либо этот выход можно принудительно программно включить при работе от батареи из прошивки или графического интерфейса.
>    [image]

Спасибо за ответ, попробывал запитать от 3.3 выхода blue pill, работает. Судя по всему стоит этот стабилизатор (https://pdf1.alldatasheet.com/datasheet-pdf/view/74628/MICREL/MIC5205-3.3BM5.html) и он будет давать 3.3 Вольта

Avtopolet, 2021-11-20:
> ❗️При разработке внешнего модуля вам могут сильно помочь эти инструкции и чертежи. Обратите внимание, там есть ссылки на исходники шаблона модуля в EasyEDA и файлы контуров плат. [...]
Так, согласно документации у десятипиновой гребёнки должны быть объединены земли.
[image]

Avtopolet, 2021-11-20:
> Выглядит работоспособно, если на модуле есть подтяжки

Рассмотрел модуль, лини данных и тактирования подтянуты. В принципе схема соврадает с моим модулем.

Avtopolet, 2021-11-20:
Нечаянно закрыл...

aprosvetova, 2021-11-21:
Отлично. Вам понадобятся протоборды (просто макетки по контуру модуля) или будете изготавливать сразу свои под модуль?

Avtopolet, 2021-11-22:
>
>Отлично. Вам понадобятся протоборды (просто макетки по контуру модуля) или будете изготавливать сразу свои под модуль?

Буду делать методом тёплой ламповой фрезеровки на балконе. Но протобод взял бы без использования с возвратом (приложить-посмотреть к с воей поделке), может взять(если есть в комплекте) вот эти удлиненные гребёнки.

aprosvetova, 2021-11-22:
Да, гребёнки эти трудно найти.
Подскажите, как с вами в Телеграме связаться?
Решим по поводу доставки.

aprosvetova, 2021-11-22:
UPD: Насчет гребенок ошиблась. У нас теперь гребенки стандартной длины везде используются, достать должно быть не трудно.

Но в любом случае протоборды и гребенки можем выдать, жду контакты.

Avtopolet, 2021-11-22:
Меня можно найти по этому имени пользователя @vtopolet.

okalachev, 2021-11-24:
Имею опыт работы с IMU — занимаюсь квадрокоптерами. Делал estimation ориентации с MPU-9260.

Возможно, смогу помочь.

aprosvetova, 2021-11-24:
Мы обновили 3D модели и чертежи Флиппера, теперь они соответствуют продакшн-версии, которая будет у людей на руках. На эти модели и чертежи можно ориентироваться при создании внешних модулей.

https://github.com/flipperdevices/flipperzero-3d-models

Avtopolet, 2021-11-25:
Прошу включить okalachev в команду, если это возможно.

aprosvetova, 2021-11-25:
@Avtopolet, информацию приняли, спасибо. Отредактируйте описание ишью, там можно внести ник человека.

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

okalachev, 2021-11-30:
Зарегистрировался.

Avtopolet, 2021-12-01:
Мы идём

Avtopolet, 2021-12-01:
У нас там нет случайно проблемки? Ато "билеты не выданы".

aprosvetova, 2021-12-01:
Все хорошо, мне просто нужно добраться до компа. Олегу уже выдала :) Ждите в течение часа