flipperdevices/flipper-hackathon-moscow #20: Реализация функционала взаимодействия с GSM модулем

sapoarty, 2021-11-19:
### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### Подробно опишите идею вашего проекта

Хотелось бы реализовать новый пункт меню, позволяющий осуществлять взаимодействие с GSM модулем по UART. Таким образом появляется возможность как получения звонков и смс от авторизованных номеров с последующими заданными действиями, так и возможно отправлять данные с Флиппера по заданным триггерам, типа получения радиосигнала. В будущем возможно расширение функционала для взаимодействия Флиппера с сетью интернет через подключенный модуль. 

### Готовый проект будет уметь...

Результатом будет реализация нового меню с возможностью добавлять авторизованные номера и привязывать к ним простые действия. При получении звонка и/или смс, отправляется сигнал по 433МГц с данными из смс. При получении заданного сигнала 433МГц осуществляется звонок и/или отправка смс с данными из полученного сигнала.

karasevIA, 2021-11-19:
У флиппера хватит токоотдачи батарейки на GSM?

aprosvetova, 2021-11-19:
Скажите, у вас есть опыт работы с GSM-модулями? Есть какие-то наработки?
Те модули, что я использовала, выглядели простыми поначалу, но оказались капризными и нестабильными.
Кажется, что это может быть слишком сложной задачей для пары дней Хакатона.

LionZXY, 2021-11-19:
Вот тут человек за две недели смог сделать GSM модуль:
https://www.youtube.com/watch?v=ImQ_TO92kGM

aprosvetova, 2021-11-19:
@LionZXY, это не GSM-модуль.

@sapoarty подготовьте, пожалуйста, принципиальную схему для модуля, присоединяемого к гребёнке. Тогда одобрим проект и заранее пришлём вам протоборды для удобного прототипирования, чтобы выглядело хорошо и вставлялось удобно.

Avtopolet, 2021-11-19:
> Вот тут человек за две недели смог сделать GSM модуль на флипппере: https://www.youtube.com/watch?v=ImQ_TO92kGM

Может GPS?

LionZXY, 2021-11-19:
> > Вот тут человек за две недели смог сделать GSM модуль на флипппере: https://www.youtube.com/watch?v=ImQ_TO92kGM
> 
> Может GPS?

Да, я перепутал и ошибся

aprosvetova, 2021-11-20:
❗️При разработке внешнего модуля вам могут сильно помочь эти инструкции и чертежи. Обратите внимание, там есть ссылки на исходники шаблона модуля в EasyEDA и файлы контуров плат. 

https://docs.flipperzero.one/ru/development/external-module/guide

https://docs.flipperzero.one/ru/development/ext-modules

aprosvetova, 2021-11-24:
Мы обновили 3D модели и чертежи Флиппера, теперь они соответствуют продакшн-версии, которая будет у людей на руках. На эти модели и чертежи можно ориентироваться при создании внешних модулей.

https://github.com/flipperdevices/flipperzero-3d-models

aprosvetova, 2021-11-30:
К сожалению, не дождались ответов.

closed 2021-11-30 by aprosvetova