flipperdevices/flipper-hackathon-moscow #22: UART/SPI/I2C logger

MVGavrilov, 2021-11-19:
### Подробно опишите идею вашего проекта

Хотел бы сделать приложение, позволяющее полключить flipper к шинам данных UART, i2c, SPI, и вести логирование передаваемых данных на sd-карту в черовекочитабельном виде. Часто бывает полезным при отладке устройств, особенно если данными устройства обмениваются редко (оставил на ночь флипер логировать, утром посмотрел результаты и понял почему всё опять упало)

### Готовый проект будет уметь...

Записывать логи передачи данных по i2c/uart/spi (в случае с последним - указывать, на каком из gpio пинов изменилось состояние для определения выбранного slave) на карту памяти.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

zhovner, 2021-11-19:
Запись будет происходить на SD-карту?

MVGavrilov, 2021-11-19:
Да

DrZlo13, 2021-11-19:
Чуть-чуть реализовано тут https://github.com/flipperdevices/flipperzero-firmware/pull/831, если поможет

MVGavrilov, 2021-11-19:
> Чуть-чуть реализовано тут [flipperdevices/flipperzero-firmware#831](https://github.com/flipperdevices/flipperzero-firmware/pull/831), если поможет

Круто, спасибо!

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации