flipperdevices/flipper-hackathon-moscow #23: Обнаружение, подключение и управления дистанционно управляемыми вибромассажерами.

MikeRebel, 2021-11-19:
### Подробно опишите идею вашего проекта

Сделать функционал по управлению с устройства дистанционно управляемыми вибромассажерами. 

### Готовый проект будет уметь...

Можно подключить вибромассажер, увеличить и уменьшить силу вибрации, переключить режим.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

zhovner, 2021-11-19:
Все нормальные BLE устройства используют криптографию. Вы реализовывали подобно уже?

MikeRebel, 2021-11-19:
Немного непонятно, почему вы решили, что это проблема? Я не про взлом устройства, а про управление им хочу сделать ПО. Опыта нет

aprosvetova, 2021-11-19:
Да, протоколы многих игрушек отреверсены, и при желании их можно реализовать на Флиппере.
Только имейте в виду, что наш BLE пока не имеет удобных высокоуровневых API для взаимодействия с другими устройствами, кроме собственного протокола RPC.

vvzvlad, 2021-11-23:
@MikeRebel Можно в команду?

MikeRebel, 2021-11-23:
Конечно

vvzvlad, 2021-11-23:
@aprosvetova что надо сделать для добавления участника? Просто отредактировать первый пост?

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

vvzvlad, 2021-11-30:
Зарегистрировался.

MikeRebel, 2021-12-01:
Привет. 
Я не успеваю поучаствовать, от нас будет 1 человек.

aprosvetova, 2021-12-01:
Договорились, спасибо!