flipperdevices/flipper-hackathon-moscow #25: Лайтовый Telegram клиент для флиппера

Astrrra, 2021-11-19:
### Подробно опишите идею вашего проекта

Клиентское приложение, принимающее данные от кастомного Telegram клиента по Bluetooth и позволяющее общаться текстом 

### Готовый проект будет уметь...

Будет:

- Показывать список из последних N чатов
- При выборе чата показывать последние М собщений и поле ввода текста
- Получение новых сообщений и отправка сообщений с флиппера при помощи его клавиатуры
(N и M больше 3 и меньше 100)

Не будет:
- Показывать стикеры, эмодзи, картинки и прочие вложения
- Звонить
- Проигрывать войсы
- Все остальное)

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@Vital7

### GitHub третьего участника

@SunnyCapt

Astrrra, 2021-11-19:
Лучше?)

aprosvetova, 2021-11-19:
Ага! Идея красивая, причем реализуемая. Мы решили подготовить RPC-сообщение для связи сторонних приложений с внешним миром, поэтому сможете реализовать это дело, даже не трогая блютус.

Но понимаете ведь, что на клавиатуре Флиппера даже пробела нет?

Astrrra, 2021-11-19:
Можно сделать так что приложение какое-то сочетание букв будет считать пробелом

Ну или свою написать но это довольно больно будет я думаю
Но в теории есть вариант использовать D-pad для выбора букв и символов по дереву, так что в прииииинципе...

aprosvetova, 2021-11-19:
Ну можете _ взять как пробел)

Astrrra, 2021-11-19:
А, там же _ есть, можно на пробел заменять при отправке

Astrrra, 2021-11-19:
> Ну можете _ взять как пробел)

Хах, одновременно

skotopes, 2021-11-19:
Шрифты у нас без юникода

aprosvetova, 2021-11-19:
При желании можно и юникодовые подключить, я без проблем рендерила хирагану на Флиппере.
Но более правильным решением видится транслитерация на стороне моста.

Astrrra, 2021-11-19:
> При желании можно и юникодовые подключить, я без проблем рендерила хирагану на Флиппере. Но более правильным решением видится транслитерация на стороне моста.

Мы так и думали сделать

Astrrra, 2021-11-21:
> 

Так, после некоторого гуглинга я поняла что не совсем понимаю как работать с RPC, особенно конкретно в случае флиппера. Про это есть где-то в доках?

Astrrra, 2021-11-21:
> Ага! Идея красивая, причем реализуемая. Мы решили подготовить RPC-сообщение для связи сторонних приложений с внешним миром, поэтому сможете реализовать это дело, даже не трогая блютус.
> 
> Но понимаете ведь, что на клавиатуре Флиппера даже пробела нет?

Так, после некоторого гуглинга я поняла что не совсем понимаю как работать с RPC, особенно конкретно в случае флиппера. Про это есть где-то в доках?

aprosvetova, 2021-11-21:
Пока нет.

Astrrra, 2021-11-21:
Окей

LionZXY, 2021-11-22:
Я буду без дела на хакатоне, смогу помочь с этим

Astrr, 2021-11-22:
> Я буду без дела на хакатоне, смогу помочь с этим

О, будет очень неплохо, спасибо!

Astrrra, 2021-11-22:
> Я буду без дела на хакатоне, смогу помочь с этим
 
О, это будет очень полезно, спасибо!

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

SunnyCapt, 2021-11-30:
Прошел регистрацию

Astrrra, 2021-11-30:
Зарегалась

Vital7, 2021-11-30:
Тоже зарегалась

aprosvetova, 2021-11-30:
Умнички, всем отправила билеты.

Astrrra, 2021-12-01:
Тест