flipperdevices/flipper-hackathon-moscow #26: Игра Pong

code-de, 2021-11-19:
### Подробно опишите идею вашего проекта

Реализация игры Pong на внутреннем экране Флиппера с режимом игры с ИИ и возможно мультиплеером. Обязательно добавим поездатый саундтрек.

### Готовый проект будет уметь...

В игре с ИИ планируем реализовать 3 уровня сложности, в зависимости от которого будет меняться (поэкспериментируем, как лучше играется на маленьком экране): уровень ошибок ИИ, скорость мячика и/или размер платформы. Учёт и сохранение high score'ов для каждого из уровней сложности. Юзание вибромоторчика при проигрыше для иммерсивного ощущения своей ничтожности.

Опционально, если хватит времени, очень хотели бы реализовать мультиплеер для игры с другими Флипперами через BLE или радио диапазон.
Попробовали бы также сделать быстрый синхрон двух Флипперов через NFC чтобы быстро пэйрить девайсы касанием и сразу погнать играть.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@DonPerenion

### GitHub третьего участника

@CrazyTaksa

aprosvetova, 2021-11-19:
> для иммерсивного ощущения своей ничтожности

Ah, I see you're a man of culture as well.

>Опционально, если хватит времени...

В этом разделе важно четко описать то, что будет считаться критерием законченности проекта. Именно по этому критерию мы будем судить, справились вы с задачей или нет. Поэтому опциональное, пожалуйста, перенесите в первый блок.

А игру давайте делать)

code-de, 2021-11-20:
> Ah, I see you're a man of culture as well.
>
>
> В этом разделе важно четко описать то, что будет считаться критерием законченности проекта. Именно по этому критерию мы будем судить, справились вы с [...]

Спасибо, Аня!🤞 Блоки отредактировал)

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

code-de, 2021-11-30:
Врегался

aprosvetova, 2021-11-30:
@donperenion, вы тоже зарегались, да?

DonPerenion, 2021-11-30:
> @DonPerenion, вы тоже зарегались, да?

Да, зарегался!

CrazyTaksa, 2021-12-01:
Готово!