flipperdevices/flipper-hackathon-moscow #27: Работа с ключами ds199х

Erop1928, 2021-11-19:
### Подробно опишите идею вашего проекта

Хотел бы реализовать работу с внутренней памятью ключей ds1992/1995 (и прочих, данной серии если смогу их достать)

### Готовый проект будет уметь...

Читать внутреннюю память ключа
Эмулировать ключ 
Записывать в память ключа 

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@KateSuleymanova

### GitHub третьего участника

(будет еще человек, но он позже вышлет свой гит)

aprosvetova, 2021-11-19:
Очень полезная и нужная идея. Если получится реализовать, смержим в основную прошивку. 

Когда третий человек объявится, отредактируйте, пожалуйста, пост.

Erop1928, 2021-11-21:
Человек объявился

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

Erop1928, 2021-11-30:
Я прошел регистрацию

KateSuleymanova, 2021-11-30:
Я прошла регистрацию

ma3a3, 2021-11-30:
Я прошел регистрацию