flipperdevices/flipper-hackathon-moscow #28: Вольтметр, анализатор протоколов быстрого заряда

Serga74, 2021-11-19:
### Подробно опишите идею вашего проекта

 модуль тестирования блока питания: напряжения,
протокола заряда (PD (с PPS), QC2.0, QC3.0, QC4.0, Huawei SCP, SSCP, FCP, Samsung AFC),
пульсация напряжения (осцилограф) подобие (AVHzY CT3, FNB48)
тестирование кабелей usb

### Готовый проект будет уметь...

Отображение и логирование ифнрмации о напряжениии имеющихся протоколах, оформленное отдельном модулем, с возможностью вывода информации на ПК или телефон (режим осцилографа), проверка сопротивления кабеля USB

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

aprosvetova, 2021-11-19:
Идея хорошая, но сложная со стороны железа.
Мы не сможем обеспечить никакими инстурментами для прототипирования железа на Хакатоне, поэтому, пожалуйста, подготовьте модуль заранее.

Покажите принципиальную схему (хотя бы примерную), тогда мы одобрим проект и пришлём вам наши фирменные протоборды для удобного прототипирования.

aprosvetova, 2021-11-20:
❗️При разработке внешнего модуля вам могут сильно помочь эти инструкции и чертежи. Обратите внимание, там есть ссылки на исходники шаблона модуля в EasyEDA и файлы контуров плат. 

https://docs.flipperzero.one/ru/development/external-module/guide

https://docs.flipperzero.one/ru/development/ext-modules

aprosvetova, 2021-11-24:
Мы обновили 3D модели и чертежи Флиппера, теперь они соответствуют продакшн-версии, которая будет у людей на руках. На эти модели и чертежи можно ориентироваться при создании внешних модулей.

https://github.com/flipperdevices/flipperzero-3d-models

aprosvetova, 2021-11-30:
К сожалению, не получили никаких ответов.

closed 2021-11-30 by aprosvetova