flipperdevices/flipper-hackathon-moscow #29: Ethernet ping-тестер

karasevIA, 2021-11-19:
### Подробно опишите идею вашего проекта

Плата расширения с контроллером W5500 для работы с Ethernet

### Готовый проект будет уметь...

Устройство будет использоваться для проверки наличия интернета путем (ping 8.8.8.8) или определения маски локальной сети (при наличии DHCP).

На экран будет выводиться привычный ответ на команду пинг, или в случае локальной сети IP-адрес и маска

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@sht4

aprosvetova, 2021-11-19:
Восхитительно, уже хочу себе такую штуку! Как и во всех проектах с железными модулями, просим вас подготовить и показать хотя бы примерную принципиальную схему заранее. После этого одобрим проект и пришлем вам наши протоборды, чтобы модуль получился удобный и целиком втыкался в гребенку.

aprosvetova, 2021-11-20:
❗️При разработке внешнего модуля вам могут сильно помочь эти инструкции и чертежи. Обратите внимание, там есть ссылки на исходники шаблона модуля в EasyEDA и файлы контуров плат. 

https://docs.flipperzero.one/ru/development/external-module/guide

https://docs.flipperzero.one/ru/development/ext-modules

karasevIA, 2021-11-20:
[image]
Если упростить то так, саму схему с обвязкой и прочим выдадим сегодня-завтра, как закончим. Занимать планируем одну линейку пинов.

В целом удобнее было бы воспользоваться готовым 3.3В выходом, но ради него тянуть плату до второй линейки не хочется. Пожертвуем оптимальности в угоду лаконичности шилда

aprosvetova, 2021-11-20:
Сталкивалась с тем же, когда делала кустарный модуль ESP8266. Лаконичность — круто, но такой модуль можно случайно вставить не той стороной, и каждый раз об этом думать нервно. 

Очень рекомендуем сделать полноценный модуль, во всю ширину. Если нужно, пришлем протоборды, чтобы было проще. И LDO не понадобится)

karasevIA, 2021-11-20:
Мы планируем сделать модуль с корпусом, такая ошибка вряд-ли случится
Нам так же хотелось бы чертежи или модели корпуса, примерная идея на картинке 
[image]

sht4, 2021-11-21:
Схему не компилировал. Просто составил без main пока.
Там возможно SPI преобразователь уровней надо и что-то еще. На свежую голову посмотрю

1) https://github.com/sht4/SHMTC/blob/master/Flipper_ETH/Connectors.pdf
2) https://github.com/sht4/SHMTC/blob/master/Flipper_ETH/Power.pdf
3) https://github.com/sht4/SHMTC/blob/master/Flipper_ETH/Shield.pdf

sht4, 2021-11-22:
Вот схема нашего устройства.
https://github.com/sht4/SHMTC/blob/master/Flipper_ETH/Flipper_ETH.pdf

karasevIA, 2021-11-22:
@aprosvetova, что-то еще нужно?

aprosvetova, 2021-11-23:
Собрала рекомендации по вашей схеме от коллеги:
- Пины 4 и 5 коннектора X1 на землю
- SCLK ошибочно замкнут на 3V3 рядом с R16, осторожно
- По даташиту W5500, C22 должен быть 10 nF
- Его же пины 38-42 не подтянуты к земле _(впрочем я в даташите вижу NC)_

В целом давайте делать! Нужны протоборды?

karasevIA, 2021-11-23:
Спасибо! 
Исправим
Прерывания использовать не планировали, именно для нашего функционала не понадобятся. Глобально в модуле не помешает - добавим.

Протоборт не нужен, спасибо
А вот старенький корпус пригодился бы, мы бы тогда смогли какой-нибудь корпус напечатать

aprosvetova, 2021-11-23:
> Прерывания использовать не планировали
Понимаю. У нас просто сложилась практика заводить всё, что можно. Кто знает, какие еще реализуемые программно задачи появятся) 

Корпус, к сожалению, предложить не можем, но есть 3D-модели: http://github.com/flipperdevices/flipperzero-3d-models
Сегодня-завтра обновим их на актуальные.

sht4, 2021-11-23:
> * Пины 4 и 5 коннектора X1 на землю
> * SCLK ошибочно замкнут на 3V3 рядом с R16, осторожно. Клок по идее напрямую из Флиппера заводится, без резистора.
> * По даташиту W5500, C22 должен быть 10 nF
> * Его же пины 38-42 не подтянуты к земле _(впрочем я в даташите вижу NC)_
> * На Флиппер не выведен REST
> * На Флиппер не выведен INT (возможно, захотите обрабатывать пакеты по прерываниям)

1) Исправил - пины 4 и 5 коннектора X1 на землю.
2) Исправил -  SCLK ошибочно замкнут на 3V3 рядом с R16, осторожно. Клок по идее напрямую из Флиппера заводится, без резистора.
3) Исправил -  по даташиту W5500, C22 должен быть 10 nF

4) NC Его же пины 38-42 не подтянуты к земле (впрочем я в даташите вижу NC). По референсной схеме они оставлены в воздухе. Altium при преобразовании в pdf убрал крестики, которые закрывают пины, как неиспользуемые. В этой референсной схеме они тоже в воздухе висят:
https://docs.wiznet.io/img/products/w5500/w5500_sch_v110_use_trans_.pdf

5) Исправил -  на Флиппер не выведен REST.
6) Исправил - на Флиппер не выведен INT (возможно, захотите обрабатывать пакеты по прерываниям).

Ссылка на исправленую схему:
https://github.com/sht4/SHMTC/blob/master/ETH_Flipper/Flipper_ETH.pdf

aprosvetova, 2021-11-23:
Класс!

sht4, 2021-11-23:
> Класс!

Вы тоже отправляете на проверку после правок?

aprosvetova, 2021-11-23:
Да не, дальше уже сами.
Идею вашу одобрили ещё до правок, ждём вас на Хакатоне.

Skorpionm, 2021-11-23:
А вот 32 пин (SCn), как и 37 (Reset) подтянуть в 3.3, иначе ништяков нахватаетесь

sht4, 2021-11-23:
> А вот 32 пин (SCn), как и 37 (Reset) подтянуть в 3.3, иначе ништяков нахватаетесь

Благодарю, поправил!
Еще добавил подтяжку на 36 пин - внешнее прерывание.

thecpcat, 2021-11-23:
Может, вместо 8.8.8.8 пинговать default gateway из DHCP? А то инет-то не всегда есть.

karasevIA, 2021-11-23:
Если адрес выдан, то уж пинг добежит до роутера. Модуль нацелен на хотя бы минимальную полезность, в работе еще не было необходимости пинговать default gateway, если честно. Всегда нужно проверить как раз то что указал в описании. Если считаете функцию полезной - у вас всегда будет возможность добавить ее самому, все исходники будут открыты конечно же

Skorpionm, 2021-11-24:
36 пин INT это ВЫХОД с W5500 и резистор там некчему

sht4, 2021-11-24:
> 36 пин INT это ВЫХОД с W5500 и резистор там некчему

Тогда у меня есть ряд вопросов.
Можно их обсудить в чате или личке тг?
Там по разной схеме разная разводка одной и той же ноги w5500

Skorpionm, 2021-11-24:
ну давайте так. 
1. я не претендую на единственно верное решение, вы вправе делать так как считаете нужным. 
2. вы хотите использовать голую микросхему и набор деталей для для прототипа? почему не использовать готовый модуль на W5500? 99.9% вопросов отпадут сами собой и не нужно не чего паять  лишнего.

aprosvetova, 2021-11-24:
Мы обновили 3D модели и чертежи Флиппера, теперь они соответствуют продакшн-версии, которая будет у людей на руках. На эти модели и чертежи можно ориентироваться при создании внешних модулей.

https://github.com/flipperdevices/flipperzero-3d-models

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

karasevIA, 2021-11-30:
Все в порядке

karasevIA, 2021-11-30:
Регистрацию прохожу сейчас

sht4, 2021-11-30:
> ### Ждём ваше финальное решение
>
> **Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.
>
> **Если изменилось число участников**, напишите здесь, кого нужно исключить.
>
> **Если вы [...]

Прошёл регистрацию.