flipperdevices/flipper-hackathon-moscow #30: Портативный спуфер карт с магнитной полосой

dunaevai135, 2021-11-19:
### Подробно опишите идею вашего проекта

Плата расширения со сканером магнитных карт и эмулятор на H мосте ( http://samy.pl/magspoof/ ). Магнитные карты до сих пор используются в отелях, старых СКУДах, и как бонусные карты.

### Готовый проект будет уметь...

Считывать и эмулировать (для понимания как это выглядит см. проект SamyKamkar http://samy.pl/magspoof/ ) магнитные карты.

На экран будет выводиться интерфейс аналогичный работе с RFID картами.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

aprosvetova, 2021-11-23:
Класс! Давайте делать.

### По поводу железных проектов
Мы не сможем обеспечить никакими инстурментами для прототипирования железа на Хакатоне, поэтому, пожалуйста, подготовьте модуль заранее.

Покажите принципиальную схему (хотя бы примерную), тогда мы одобрим проект и выдадим вам наши фирменные протоборды для удобного прототипирования, если понадобятся.

При разработке внешнего модуля вам могут сильно помочь эти инструкции и чертежи. Обратите внимание, там есть ссылки на исходники шаблона модуля в EasyEDA и файлы контуров плат.

https://docs.flipperzero.one/ru/development/external-module/guide

https://docs.flipperzero.one/ru/development/ext-modules

dunaevai135, 2021-11-23:
Примерная схема.

[image]

В целом подойдёт любой H-мост, такой уже заказал.
Ридер нужен маленький с TTL выводом (например https://www.alibaba.com/product-detail/Mini-portable-12-3-tracks-usb_60679900708.html?spm=a2700.details.0.0.68a313bcLGb05q) но будем подключать тот который успеем купить в РФ (советы?), он будет либо RS-232 либо PS2Keyboard, оба можно скормить STM. (теоретически возможно сделать на ацп и считывающей головке, но не за несколько дней)

фирменные протоборды хочется, чтоб это не было всё на соплях, но имеет ли смысл их пересылать их в Питер?
Telegram: dunaevai135

Skorpionm, 2021-11-24:
что то мне подсказывается что 4,5 ноги должны быть тоже на земле, тем более если используете левую пару драйвера

aprosvetova, 2021-11-24:
Мы обновили 3D модели и чертежи Флиппера, теперь они соответствуют продакшн-версии, которая будет у людей на руках. На эти модели и чертежи можно ориентироваться при создании внешних модулей.

https://github.com/flipperdevices/flipperzero-3d-models

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

AYaro, 2021-11-30:
Зарегистрировался

dunaevai135, 2021-11-30:
зарегистрировался

dunaevai13579+hackfl@gmail.com