flipperdevices/flipper-hackathon-moscow #31: RF-чат

forthe94, 2021-11-20:
### Подробно опишите идею вашего проекта

Чат на RF-передатчике.

### Готовый проект будет уметь...

Можно будет обмениваться сообщениями. Связь будет работать между модулями настроенными на одной частоте. Будет возможность включить режим шифрования, чтобы разобрать сообщение можно было только имея ключ.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

aprosvetova, 2021-11-20:
Кажется, идея дублирует #13?

forthe94, 2021-11-20:
Да точно

closed 2021-11-20 by forthe94

lomalkin, 2021-11-29:
Реализация: https://github.com/flipperdevices/flipperzero-firmware/pull/851