flipperdevices/flipper-hackathon-moscow #32: Flipper рядом

mega8bit, 2021-11-20:
### Подробно опишите идею вашего проекта

Хотел бы сделать приложение, которое транслирует 433МГц сообщение или имя Флиппера, в то время как другие дельфинщики будут видеть, что недалеко от них находится человек с Флиппером.

Подробно:
- Человек включает транслирующее\передающее приложение и идет гулять или на мероприятие
- Раз в заданное время посылается информация
- Если рядом появляется флиппер, то его уникальное имя высвечивается на дисплее отправителя (в идеале еще и издается вибрация или звук) и на дисплее получателя
- Возможно, люди знакомятся или обмениваются приветствиями и восхищаются Флипперами друг друга

### Готовый проект будет уметь...

Работающее приложение состоит из экрана, в котором отображается список имен Флипперов, находящихся рядом с владельцем Флиппера.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@dwdraugr

aprosvetova, 2021-11-20:
Выглядит так, будто это часть проекта #13, то есть идея повторяется. Может, стоит присоединиться к ребятам и вместе сделать чат?

closed 2021-11-21 by mega8bit

lomalkin, 2021-11-29:
Вместо #13: https://github.com/flipperdevices/flipperzero-firmware/pull/851