flipperdevices/flipper-hackathon-moscow #33: MIDI-контроллер/монитор

okalachev, 2021-11-21:
### Подробно опишите идею вашего проекта

Проект заключается в том, чтобы превратить Флиппер в MIDI-контроллер и MIDI-монитор. То есть в контроллер для управления музыкальными программами и в инструмент для мониторинга событий (нот), которые эти программы генерируют.

Флиппер будет подключаться к компьютеру или смартфону по Bluetooth используя [MIDI over BLE](https://www.midi.org/specifications-old/item/bluetooth-le-midi) либо (опциально) по USB.

Нажатие и отпускание кнопок Флиппера будет генерировать настраиваемые MIDI-события соответствующих нот. Также (опционально) возможен триггер MIDI-событий по каким-либо внешним источникам, например, по ИК-порту.

Таким образом, используя музыкальные программы (DAW), на Флиппере (или на пульте от телевизора!) можно будет сыграть простую мелодию, барабанный рисунок или тригерить включение того или иного сэмпла/лупа. 

А используя несколько Флипперов, можно будет даже создать нечто вроде Флипперного оркестра.

Функция мониторинга будет отображать нажимаемые в DAW ноты, их инструмент, громкость и прочие параметры, а также логировать их. Например, в .midi-файл или в специальное меню.

### Готовый проект будет уметь...

Подключение по MIDI over BLE к компьютеру или смартфону. Отображение в системе как MIDI-устройства. Мониторинг MIDI-событий на экране Флиппера. Отправка MIDI-событий по нажатию кнопок на флиппере, 

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

aprosvetova, 2021-11-23:
Идея достаточно объемная, требует реализации MIDI как протокола. Также нужно работать с BLE, что у нас сейчас очень тяжело, нормальных API нет.

Получается, что вы потратите очень много времени и сил на проект, который даже если получится, позволит сыграть лишь 6 нот (как контроллер), ведь больше у Флиппера кнопок нет.

Для хакатонов лучше наоборот выбирать яркие и легкореализуемые идеи, чтобы точно справиться и остаться довольным.

closed 2021-11-23 by aprosvetova

okalachev, 2021-11-23:
1. Проект НЕ требует реализации MIDI, поскольку для работы с MIDI уже существует огромное количество готовых библиотек. Кроме того, существуют готовые библиотеки для работы с MIDI-BLE, например [BLE-MIDI для Arduino](https://github.com/lathoub/Arduino-BLE-MIDI). И реализовать MIDI-BLE-контроллер совсем не сложно, вот, например, подобный проект с использованием ESP32: [midi-bluetooth клавиатура на esp32](https://habr.com/ru/post/513494/) (не знаю, как у Флиппера с поддержкой Arduino-библиотек).

    Я как раз специально выбрал проект, который я смогу реализовать за короткое время. У меня есть опыт работы как с MIDI, так и с BLE.

2. MIDI-контроллеры не обязательно требуют огромного количества кнопок, потому они не просто играют ноты, они триггерят MIDI-события, на которые в DAW можно настроить что угодно: как воспроизведение нот, так и активацию лупов, переключение эффектов, управление темпом и прочее. Плюс игра на виртуальных барабанах не требует большого количества нот. Также можно триггерить событие с внешних источников, что тоже кажется интересным.