flipperdevices/flipper-hackathon-moscow #34: Игра Arkanoid

Kaedone, 2021-11-21:
### Подробно опишите идею вашего проекта

Реализация игры Arkanoid на внутреннем экране Флиппера стандартный режим с несколькими готовыми уровнями (мы рассчитываем на примерно 21, с использованием встроенной во Флиппер крестовины.

Возможно, если нам хватит времени мы бы хотели бы реализовать редактор уровней, с возможности сохранения данных уровней во внутреннюю память. 
Хотим добавить множество различных дополнений (это те штучки которые падают после разбивания кирпичиков.
Так же мы туда хотим впихнуть саундтрек в старых добрых традициях кейгенов и 8битной музыки.

В последствии (если, точнее когда, все получится) мы хотим сделать еще и редактор уровней на ПК, так как редачить уровни с непосредственно flipper`a не особо удобно (хотя юзеры должны страдать, не так ли?).

### Готовый проект будет уметь...

В игре будет 21 уровень, уровни будут делиться на 3 сложности, так же после каждых 7 уровней у нас будет битва с боссом (уровень такой же как и  стандартный, но с особыми усилениями, специальной музыкой и отдельными декорациями). Так же мы надеемся, что в последствии можно будет сделать функцию передачи уровней по NFC, т.е. мы сможем сделать уровень на своем flipper`e, а после этого перекинуть его своему другу, что бы он мог страдать проходя его. 

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

aprosvetova, 2021-11-23:
Вам бы переманить одного человека из #26, чтобы было честнее.
А то у них проект проще, а команда в три раза больше)

Kaedone, 2021-11-24:
> Вам бы переманить одного человека из #26, чтобы было честнее. А то у них проект проще, а команда в три раза больше)

а можно добавить участников в комманду?
(у меня появились еще 2 личности, но не говорите никому об этом)

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

Kaedone, 2021-12-03:
Нужно срочно добавить участника
@0xaaa
Есть возможность сделать это?

aprosvetova, 2021-12-03:
К сожалению, уже нет. Регистрация закрыта, а все лимиты по участникам и так превышены.

0xaaa, 2021-12-03:
да, подтверждение не приходит

Kaedone, 2021-12-03:
Нужно произвести замену вместо меня идёт @0xaaa