flipperdevices/flipper-hackathon-moscow #35: Эмулятор chip-8

dwdraugr, 2021-11-21:
### Подробно опишите идею вашего проекта

Реализация эмулятора [CHIP-8](https://en.wikipedia.org/wiki/CHIP-8) на базе флиппера, используя доступные кнопки для управления в играх. Как и полагается для эмулятора, в нём можно будет запускать программы и игры для него, сохранять состояния и (опционально) смотреть на отладочный дамп.

### Готовый проект будет уметь...

Готовая программа будет способна запускать программы и игры для chip-8 (pong, tetris, etc..), адекватно ими управлять, используя имеющиеся элементы управления, а так же сохранять и загружать состояния программ.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@mega8bit

aprosvetova, 2021-11-23:
Ждём вас!

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

aprosvetova, 2021-11-30:
Спасибо, ждём регистрацию от @mega8bit.