flipperdevices/flipper-hackathon-moscow #36: Flipper IRtag

Rebell-Leader, 2021-11-21:
### Подробно опишите идею вашего проекта

Хотим сделать расширение для Flipper'а для игры в LaserTag с отдельным детектором попаданий и возможностью разбивки игроков на команды с подсчетом статы.

### Готовый проект будет уметь...

Наше расширение будет представлять из себя 3d-печатный пистолет для  Flipper'а и отдельный детектор попаданий с креплением на голове, связывающийся с flipper по беспроводному протоколу. Само приложение на flipper будет поддерживать разбивку игроков поблизости на команды, сбор статы и настройки матча. Момент попадания будет сопровождаться светозвуковой индикацией на приемной части и самом флиппере.
Definitions of done: 
* Выстрел из одного флиппера на детектор, привязанный к другому и успешная регистрация попадания.
* Индикация попадания.
* Запись статы матча.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@gasiki

### GitHub третьего участника

@CheHumbleProgger

Rebell-Leader, 2021-11-21:
* Gun scheme
[image]

* Detector scheme
[image]

* Simple optics
[image]

alexanderkulagin, 2021-11-22:
Выглядит очень круто, но кажется будет много сложностей с механикой. У вас есть опыт в таких делах?

gasiki, 2021-11-22:
С блютосом на ESP опыта мало, с остальным много, я собственно увлекаюсь всяким IOT и аддитивными технологиями. Занимаюсь 3д печатью с 2013 года, первую прушку собрал из резанного кита из китая , на рампсе1.3 и деталях от мфу... ESP 32 есть в наличии в мастерской на dev-kit'ах типа node-mcu. Так что если будет понятно что мы участвуем большую часть механик я смогу поверить на своём железе, останется только перенести всё на флиппер. Ну и надеяться что выложенные модели флиппера достаточно ровные чтобы по ним проектировать пистолет.

Skorpionm, 2021-11-22:
Не будет это работать, у IR диодов широкий угол излучения и их 3, да ещё и под наклоном к нужному направлению излучения (дуло пистолета) задолбаетесь фокусировать

gasiki, 2021-11-23:
У нас и нет желания его фокурировать , мы хотели использовать своеобразную диафрагму (как раз как ствол) с таким стволом длиной 5 см и диаметром 3мм , мы моем получить пятно 30 см на 5ти метрах , чего в общем то достаточно для перестрелки в офисе . Остаётся вопрос только с достаточностью яркости.

gasiki, 2021-11-23:
Ну и если оно не заведётся , то вставить 3мм диод в этот же ствол дело 5ти минут.

gasiki, 2021-11-23:
Ну а если хватит яркости диода но не получится поставить ствол достаточно напротив пучка , мы можем сделать ствол длинной 10 см при диаметре 3 мм и получить те же 30см пятна на 5ти метрах , но уже от линейного а не точечного источника.

aprosvetova, 2021-11-23:
Идея эпичнейшая, обязательно найдет поклонников среди юзеров Флиппера, не забрасывайте её.

Только для Хакатона это слишком объемный проект, который точно не осилить за пару дней.