flipperdevices/flipper-hackathon-moscow #37: Классическая игра в Змейку (Snake Xenzia)

Ximerixx, 2021-11-21:
### Подробно опишите идею вашего проекта

Хотим сделать максимально похожую игру змейка по образу той которая есть в каждой нокии

### Готовый проект будет уметь...

Наша игра будет уметь:
- Два режима игры, как в оригинале: Классический и готовые лабиринты
- Большие и маленькие яблоки
- Настраиваемая сложность
- Оценка счета (Новичок, Продвинутый, Мастер) по итогам игры
И скорее всего, мы будем делать её похожей на оригинал

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@RespectHz

LionZXY, 2021-11-22:
Дубликат
https://github.com/flipperdevices/flipper-hackathon-moscow/issues/4

aprosvetova, 2021-11-23:
Змейка уже есть.

closed 2021-11-23 by aprosvetova