flipperdevices/flipper-hackathon-moscow #38: Анализатор спектра

forthe94, 2021-11-21:
### Подробно опишите идею вашего проекта

Сделать модуль который поможет в отладке RF-девайсов работающих в субгигагерцовом диапазоне.
Будет полезен при разработке RF-девайсов, чтобы быстро понять выдает ли девайс что-то в эфир и на каких частотах.

### Готовый проект будет уметь...

Модуль сможет строить АЧХ для выбранного диапазона частот. Будут доступны:

1. Меню настроек RF трансивера
2. Меню выбора диапазона отображаемых частот
3. Выбор параметров отображаемого графика(максимум или плавающее среднее за последние N секунд)
4. График АЧХ(график можно будет очистить)

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@vostrenkov

### GitHub третьего участника

@ac1dloop

skotopes, 2021-11-21:
Если будет использоваться CC1101 то хотелось бы увидеть сначала верификацию в цифрах и формулах.

Смена частоты и подсчет RSSI занимают достаточно много времени, плюс время на ожидание шины и общение с чипом. И из примерных прикидок ничего толкового не получится, либо это будет чудовищно медленно.

forthe94, 2021-11-22:
https://elim.ru/maketnye-platy-instrumenty-ocenki/aksessuary-komplektov-etalonnyh-dizaynov-i-specialnogo-primeneniya/msp-sa430-sub1ghz-texas-instruments/
Вот например анализатор сделаный на cc1101.
Делал трансивер на s2lp и не помню чтобы там тратилось много времени на смену частоты и измерение RSSI. Не знаю сколько времени уйдет на ожидание шины. Но если секунд за 5-10 можно будет получить АЧХ модуль был бы довольно полезным.

Skorpionm, 2021-11-22:
Что таким анализатором который по 10 сек сканит 1 частоту вы собираетесь сканировать? И как вообще за вменяемое колличество времени вы собираетесь просканить весь диапазон чтоб вообще понять на какой частоте вести измерение АЧХ
Если на 1 шаг измерения частоты и получение RSSI надо минимум 5 мс. Максимальный Фильтр пропускания может быть 850КГц, но реально при таком фильтре проходит около 600КГц

forthe94, 2021-11-22:
Например известна примерная частота работы девайса, но неизвестна несущая и девиация. Можно настроить флиппер на измерение сначала на широком диапазоне, а потом сужать и получить довольно точные значения.
За 5-10 секунд я планирую получить уже распределение rssi по частотам.
Ну вот 600 КГц с шагом в 10 КГц это 60 измерений по 5мс, получается 300мс

Skorpionm, 2021-11-22:
Секундочку. 600КГц это шин на канала который пропустит фильтр допустим для сс1101 будет без разницы 433.400 ... 434.00MHz  сигнал, этот фильт можно скрутить минимум на 37КГц (по-моему не меньше)  время передачи какого нибуть брелка, в FM диапазоне (раз уж дело зашло о девиации) буквально 3-5 посылок и тишина , это не больше 100мс, дальше тишина.
И 2 вопрос как вы собираетесь рассчитать дивиацию с минимальной шириной фильтра 37КГц , не правильно измеренным RSSI на этой частоте, ТК пуль передает сигнал и уровень сигнала постоянно пляшет и не понятно в какой момент вы измерите RSSI, и отсутствием постоянно передаваемого сигнала?.  Ну может я конечно что-то не понимаю.

forthe94, 2021-11-22:
Самое простое брать максимум RSSI за всё время измерений. Вообще анализаторы для 433 МГц так и работают. Придётся пару раз нажать на кнопку передатчика, но всплески маскимумов мы увидим. До 58КГц можно уменьшить ширину канала. Где-то с шагом в 100 КГц можно тогда уже вполне нормально измерять. Точно конечно не измерить девиацию, но прикинуть можно. Увидеть что девайс шлёт и примерно на каких частотах.

vostrenkov, 2021-11-22:
Коллеги, может быть вы неправильно поняли идею проекта зацепившись за название "анализатор спектра"?
Мы не пытаемся сделать конкурента профессиональному оборудованию, наша цель - это достаточно фановое устройство, которое может оценить наличие вещания в не очень большом количестве полос и предоставить эту информацию в графическом виде (в виде спектра для удобства). 
Понятно что проводить какие-то точные измерения с помощью такой штуки бессмысленно (такой цели и нет), а вот увидеть работают ли в принципе железяки у тебя на столе вполне можно.

skotopes, 2021-11-22:
@Skorpionm для хакатона норм. 

@forthe94 @vostrenkov я аппрувлю, просто помните что на одно измерение будет уходить 5мс+ и основное ограничение будет в том сигнал должен быть постоянным чтобы его было видно.
Крайне рекомендую заранее посчитать время на один проход с разными полосными фильтрами.

Skorpionm, 2021-11-22:
Чем это будет отличаться от частотного анализатора запиленного сейчас?  Вот что мне не понятно?

forthe94, 2021-11-22:
А что он умеет и где про него можно почитать?

Skorpionm, 2021-11-22:
Видео Паша по-моему выкладывал и в предпоследней обзоре было на Хабре помоему

forthe94, 2021-11-23:
Ну хотя бы тем что будет график строиться.

Skorpionm, 2021-11-23:
ну так это переписать только вьюху отображения (вернее 2 функции и добавить 1 массив)

forthe94, 2021-11-23:
На словах вначале всегда всё просто, но обычно потом всплывает куча камней. Также мы планируем делать настройки приёмника. Можно будет выбирать шаг и диапазон частот. Это всё не звучит как написать 2 функции и добавить массив.

aprosvetova, 2021-11-23:
Планировали такое, нарисовали интерфейс, но не получилось сделать юзабельным из-за описанных выше ограничений.

Может, получится у вас?

forthe94, 2021-11-23:
> Планировали такое, нарисовали интерфейс, но не получилось сделать юзабельным из-за описанных выше ограничений.
> 
> Может, получится у вас?

Должно получится)

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

ac1dloop, 2021-11-30:
1. дело сделано
2. сообщаю

forthe94, 2021-11-30:
Зарегался.

vostrenkov, 2021-12-01:
Зарегался