flipperdevices/flipper-hackathon-moscow #39: Поисковый радио-маяк из флиппера<Тема вашего проекта>

SergeyDLevashov, 2021-11-21:
### Подробно опишите идею вашего проекта

Хочется попробовать сделать поисковый маяк на cc1101 (если я правильно понимаю, именно этот модуль у Вас используется для Sub1Ghz), который будет выдавать подряд несколько тоновых сигналов _с убывающей мощностью_ , чтобы их было слышно в обычной LPD433 рации.  Таким образом, в зависимости от того, какое количество сигналов слышно в рации, можно понять, ближе Вы к маяку или дальше. А, например, экранируя собой рацию и поворачиваясь вокруг, по количеству сигналов можно оценить направление на маяк.

### Готовый проект будет уметь...

Будет подавать сигналы на одном из каналов в частоте 433, позволяющие с помощью обычной рации локализовать его место. Если удастся, будет активироваться вызывным тоном с рации.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

lomalkin, 2021-11-21:
Что-то типа маяка для "Охоты на лис"?

skotopes, 2021-11-21:
Тест несущей уже имплементирован выглядит очень похоже

SergeyDLevashov, 2021-11-22:
Это маяк, который под рукой и который можно прицепить на что угодно, чтобы это что-то можно было найти в полевых условиях даже без любой связи кроме рации. Например найти улетевший дрон или оставленное работать оборудование, собаку, в общем, любой объект который может переместиться и потом должен быть найден, либо место, которое надо найти в однотипном рельефе.  
Это должны быть несколько раздельных и различающихся сигналов с мощностью, падающей таким образом, чтобы в зависимости от расстояния (и направления) отчетливо было слышно разное количество отличающихся писков.  
В идеале маяк должен включаться по вызывному тону от рации и возможно, подавать так же звуковые сигналы, чтобы его можно было легко найти уже находясь рядом при полном сигнале. Ну и да, можно и в охоту на лис поиграть. 

Существует такое отдельное специализированное решение, меньше размером и с бо'льшим количеством возможностей. Например tbeacon - отличный легкий маленький и удобный радиомаяк. Но не всегда он ссобой, когда нужен. Раз уж flipper позиционируется как мультитул, предположил, что схожий функционал ему может оказаться полезен.

Skorpionm, 2021-11-22:
> Тест несущей уже имплементирован выглядит очень похоже

Саш ты не прав, моячку нужны несколько сигналов с разной силой сигнала (по их количеству принятому приемником можно судить приближается или удаляешься от объекта) и мало того сигнал должен быть прерывистым и в идеале несколько тонов чтоб различить их от  шума и настроиться на нужную частоту

Skorpionm, 2021-11-22:
> Это маяк, который под рукой и который можно прицепить на что угодно, чтобы это что-то можно было найти в полевых условиях даже без любой связи кроме рации. Например найти улетевший дрон или [...]

Флипер привязывать к дрону??? Он весит как треть вес дрона.

aprosvetova, 2021-11-23:
Боюсь, дальности нашего 433 не хватит, чтобы применять такой маяк там, где он правда нужен.

closed 2021-11-23 by aprosvetova