flipperdevices/flipper-hackathon-moscow #40: Flipper-PasswordKeeper

forthe94, 2021-11-21:
### Подробно опишите идею вашего проекта

Этот модуль будет превращать флиппер в переносный менеджер паролей.

### Готовый проект будет уметь...

Флиппер в режиме bad usb сможет вводить пароли по нажатию клавиши.
На флиппер пароли можно будет скидывать как на флешку(с форматом файла определимся).
Также можно будет генерировать пароли на самом девайсе.
При этом флиппер будет разблокироваться выбранной комбинацией нажатий клавиш(что-то типа пин-кода).

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@vostrenkov

### GitHub третьего участника

@ac1dloop

lomalkin, 2021-11-21:
Хорошая идея. К слову, блокировка клавиатуры девайса по секретной комбинации джойстика уже есть в штатной прошивке.

skotopes, 2021-11-21:
Так то мы больше смотрим в сторону U2F и продолжателей его дела.

aprosvetova, 2021-11-21:
Пока это звучит точь-в-точь как штатное приложение BadUSB, просто в файлах-скриптах пароли.

aprosvetova, 2021-11-23:
Извините, слабовато для такой большой команды. Отличия от rubberducky скриптов пока всё так же непонятны(

closed 2021-11-23 by aprosvetova