flipperdevices/flipper-hackathon-moscow #41: Flipper-EUCDashboard

stz1945, 2021-11-21:
### Подробно опишите идею вашего проекта

Использование флиппера в качестве приборной панели для моноколеса.

Девайс должен быть закреплён на перчатке, показывать скорость, процент батареи и ток, а также пищать и трясти райдера за руку при достижении его (колеса) параметрами опасных величин.

Смысл в том, что при достижении максимальной скорости моноколеса, соответствующей стопроцентному заполнению ШИМа его регулятора, колесо перестаёт держать горизонт, а пилот разбивает нос (и не только) об асфальт, что весьма нежелательно.

### Готовый проект будет уметь...

- Подключаться к моноколесу по bluetooth serial
- Получать пакеты со значениями скорости, силы тока, остаточной емкости батареи
- Выводить значения на экран
- Считать скважность ШИМа регулятора из напряжения и скорости
- Пищать и жужжать вибромотором по достижении пороговых значений ШИМ

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

lomalkin, 2021-11-21:
Как моноколесник скажу, что это очень полезная штука. Разбор самих протоколов можно позаимствовать из проекта [WheelLog](https://github.com/Wheellog/Wheellog.Android). Голосую за!

skotopes, 2021-11-21:
У нас BLE, не Bluetooth classic.

schtangel, 2021-11-22:
Неверно выразился - колесо тоже BLEшное, текущие значения читаются из GATT.

Судя по заявке про другие жужжалки, высокоуровнего API для этого сейчас нет?

aprosvetova, 2021-11-25:
Автор сдался 🏳

closed 2021-11-25 by aprosvetova