flipperdevices/flipper-hackathon-moscow #42: Stack Attack

M010, 2021-11-22:
### Подробно опишите идею вашего проекта

Хотел бы сделать реализацию старой игры Stack Attack, которая была популярна на старых siemens

Вольный пересказ механики игры:
В игре доблестный рабочий старается пофиксить неоптимальную службу разгрузки коробок. Боксы падают на него с неба и он рискуя собственной жизнью должен двигать их чтобы заполнять образовавшиеся пустоты. Как только ряд заполнен он магическим образом исчезает.

### Готовый проект будет уметь...

В моей игре будет: 

- Коробки которые падают с неба/крана
- Рабочий который сможет передвигать коробки (управляется с кнопок flipper)
- В случае падения на голову рабочего он будет умирать и на этом игра закончится
- Если заполняется ряд из коробок, то он исчезает и остальные коробки спускаются на ряд ниже
- Подсчет очков
- Вывод игры на экран flipper

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

в разработке

### GitHub третьего участника

в разработке

aprosvetova, 2021-11-23:
Кайф, вспомним детство!
Скорее отмечайте аккаунты участников.

M010, 2021-11-29:
Добавил аккаунты участников!)

aprosvetova, 2021-11-30:
### Ждём ваше финальное решение

**Если вы не идёте на Хакатон**, напишите комментарий здесь и закройте issue.

**Если изменилось число участников**, напишите здесь, кого нужно исключить.

**Если вы ТОЧНО идёте на Хакатон**, каждый участник команды должен:
1. Зарегистрироваться на мероприятие [по ссылке](https://neuronspace.timepad.ru/event/1860939/)
2. Сообщить здесь, в issue, о прохождении регистрации

richardtop, 2021-12-01:
Удачи, очень интересно увидеть этот проект!
У меня была такая же идея, я ее предложил в чате:

https://t.me/FlipperHackathon2021/853