flipperdevices/flipper-hackathon-moscow #43: Флиппер-Бипер

BeanSecurity, 2021-11-22:
### Подробно опишите идею вашего проекта

Плеер DTMF мелодий. Приложение которое будет воспроизводить с динамика флипера DTMF мелодии, набранные с кнопок на экранчике (1234567890*#ABCD) либо из файлов (с форматом файла разберусь позже, наверное будет просто txt с символами 1234567890*#ABCD).

### Готовый проект будет уметь...

- Воспроизводить DTMF мелодии через встроенный динамик
- Загружать мелодии из файлов на флешке и воспроизведить при нажатии на мелодию в списке
- Воспроизводить DTMF звуки в реальном времени с клавиатуры на экране (раскладка как на картинке)
[image]

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

okalachev, 2021-11-22:
А там есть встроенный динамик?

BeanSecurity, 2021-11-22:
> А там есть встроенный динамик?

да, пьезо-пипка которая, для такой задачи она словно та самая шляпа (как раз)

Skorpionm, 2021-11-22:
Есть готовая функция notification способная воспроизводить ноты 
https://github.com/flipperdevices/flipperzero-firmware/blob/dev/applications/notification/notification-messages-notes.c

BeanSecurity, 2021-11-22:
> Есть готовая функция notification способная воспроизводить ноты https://github.com/flipperdevices/flipperzero-firmware/blob/dev/applications/notification/notification-messages-notes.c

спасибо, DTMF это двухтоновый сигнал, ноты там насколько я понял это один тон (буквально одна синусоида).
Насколько я себе представляю сейчас с берега, чтобы получить два тона одновременно мне нужно будет как то сформировать особый ШИМ сигнал (никогда не работал с микроконтроллерами и вот етим всем), так что пока это выглядит чуть посложнее чем просто переиспользовать готовый функционал с нотами, но все равно спасибо

aprosvetova, 2021-11-25:
Рекомендую рассмотреть вариант доработки музыкального плеера, который сейчас есть в plugins. Он сейчас только одну мелодию играет.

Заявку пока отправляю в резерв, у нас кончился лимит. Если кто-то сольётся и будут свободные места, пригласим.

aprosvetova, 2021-12-02:
40 из 41 участников одобренных команд зарегистрировались на мероприятие и получили билеты.
К сожалению, это даже больше, чем максимальная вместимость помещения, поэтому никого дополнительно взять не можем.

Не расстраивайтесь и не забрасывайте свою идею, пожалуйста. Мы рассчитываем, что Хакатон пройдёт хорошо, и тогда ещё не раз вернёмся к этому формату.

closed 2021-12-02 by aprosvetova