flipperdevices/flipper-hackathon-moscow #44: Реализация OBEX-клиента для ИК-порта

terge-eg, 2021-11-22:
### Подробно опишите идею вашего проекта

Хочу реализовать передачу чб картинок на флиппер со старых телефонов, например Ericsson R320s на базе протокола OBEX (ссылка на спеку ниже). Мб организовать передачу с флиппера, но надо понять, на что:)

http://www.farnell.com/datasheets/306539.pdf

### Готовый проект будет уметь...

Принимать чб картинки. Мб ставить их на заставку.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

aprosvetova, 2021-11-22:
Какой транспорт предполагается? BLE, IR, UART, USB?

terge-eg, 2021-11-22:
IR для ламповости

terge-eg, 2021-11-22:
> Какой транспорт предполагается? BLE, IR, UART, USB/другое?

IR для ламповости

aprosvetova, 2021-11-22:
Почему-то в заголовке проглядела, сорри.

У нас TSOP-75538, почитайте [даташит](https://www.vishay.com/docs/82495/tsop753.pdf).
Он работает с несущей 38 kHz, надо убедиться, подойдёт ли это для реализации OBEX.
Ну и другие ТТХ нашего приёмника посмотрите, может там фильтрации какие-то, я не шарю.

aprosvetova, 2021-11-22:
Наш ИК-ресивер не потянет.

closed 2021-11-22 by aprosvetova