flipperdevices/flipper-hackathon-moscow #46: Мультиплеерные шашки.

AlexKrudu, 2021-11-23:
### Подробно опишите идею вашего проекта

Цель проекта - реализовать мультиплеерные шашки для двух игроков на внутреннем экране Флиппера с режимом игры "против компьютера".

 Управление будет осуществляться через встроенную крестовину.

### Готовый проект будет уметь...

У игры будет 2 режима:
1) Против компьютера (опционально с несколькими уровнями сложности)
2) Против человека с другим флиппером.

Для связи с другим флиппером будет использоваться либо BLE, либо радио диапазон.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@yank0vy3rdna

aprosvetova, 2021-11-25:
Идея хорошая, но у нас пока нет свободных мест :(
Если кто-то откажется от участия, что довольно вероятно, пригласим вас. Пока резерв.

AlexKrudu, 2021-11-26:
А когда будет ясно, слился кто-то или нет?

aprosvetova, 2021-11-26:
Как только кто-то сольётся :>

После выходных всех попинаем ещё.

AlexKrudu, 2021-12-01:
Доброго времени суток? Пока никто не слился?

aprosvetova, 2021-12-01:
Нет, даже среди выданных билетов небольшое переполнение. Третьего декабря днем будет точно понятно, но скорее всего все резервные и не рассмотренные заявки будут закрыты, извините :(

Если получится классно, мы обязательно когда-нибудь повторим мероприятие.

aprosvetova, 2021-12-02:
40 из 41 участников одобренных команд зарегистрировались на мероприятие и получили билеты.
К сожалению, это даже больше, чем максимальная вместимость помещения, поэтому никого дополнительно взять не можем.

Не расстраивайтесь и не забрасывайте свою идею, пожалуйста. Мы рассчитываем, что Хакатон пройдёт хорошо, и тогда ещё не раз вернёмся к этому формату.

closed 2021-12-02 by aprosvetova