flipperdevices/flipper-hackathon-moscow #47: Холодный криптокошелёк

iDoka, 2021-11-26:
### Подробно опишите идею вашего проекта

Криптовалютный холодный кошелёк. 
Для целей хранения приватников используется микросхема типа Security Enclave, подключаемая по I2C.
Сам доступ в Security Enclave защищён секретной фразой - злоумышленник не сможет завладеть активами, имея только HW add-on либо только парольную фразу.
Сами транзакции для подписи Флиппер получает с хоста по USB-CDC.
В рамках хакатона планируем ограничиться поддержкой подписания транзакций для сети Bitcoin.

### Готовый проект будет уметь...

... получать с хоста блокчейн-транзакции, подписывать их, выдавать подписанные транзакции на хост.
(+ функционирующий HW add-on разумеется)

[image]

.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@stolpa4

### GitHub третьего участника

@malmstrem

aprosvetova, 2021-11-26:
В нашем чипе уже есть secure enclave, но его недостаточно, чтобы реализовать **честный** холодный кошелек, потому что он умеет лишь хранить ключи и шифровать/дешифровать ими данные с помощью AES-CBC.

То есть можно зашифровать приватный ключ кошелька ключом из secure enclave и каждый раз расшифровывать его, когда нужно подписать транзакцию. Это грязный вариант, потому что позволяет при желании вытащить расшифрованный приватный ключ кошелька.

Внешний secure enclave решает проблему, но получается громоздкое страшное устройство, и непонятно, зачем тогда вообще Флиппер, если независимое устройство получится меньше и таким же функциональным.

Не хотите сделать грязную реализацию, но self-contained?

P.S. Сейчас в любом случае мест нет, и даже если одобрим заявку, она уедет в резерв

iDoka, 2021-11-26:
# почему standalone хранение

У команды было несколько сомнений по поводу использования встроенного во Флиппер Customer key storage. В перечислении от более значимых к менее значимым:

1. Секьюрность. Все наслышаны о "безопасности" STM32 и если на какую-то серию STM32 нет публикации об уязвимостях, то это не означает что уязвимости не существует. В этом свете использование специализированных решений выглядит куда привлекательнее. Чипы класса **ST31H320** (Certification level: CC EAL5+) нам никто не продаст, да и если бы продали, NDA не позволило бы выложить код для работы под GPL, поэтому в результате рисёча нашли разумный компромисс - серию чипов от microchip, как пример: [ATECC608A](https://www.microchip.com/en-us/product/ATECC608A).

2. Присутствовал (и присутствует) консёрн по поводу того успеем ли что-то заделиверить на WB55 за 3 дня (когда будет реальный доступ к железу), поэтому набрейнштормили макет на bluepill + чип от microchip чтобы была возможность отлаживать до хакатона. Как задача-максимум - постараемся заимплементить self-contained.

3. Физически иная (отделимая от Флиппера) сущность: если делать хоть сколько-то малую заявку на финтех, то пользователю куда спокойнее хранить приватник в *сухом прохладном месте*, а не таскать с собой в ранце как физически неотделимую от флиппера сущность (мультисиги не рассматриваем).

# про мест нет

В [изначальном посте](https://habr.com/ru/company/flipperdevices/blog/589585/) ничего не было сказано что критерием для отбора проекта будет приоритет даты подачи заявки (а не, например, проработанность идеи или экспертиза команды), но сейчас это выглядит так 🤷‍♂️

Мы не коммитили идею до тех пор пока не было ясности сможем ли мы достать хоть какие-то релевантные задаче чипы во вменяемое время в условиях чипагеддона, чтобы участвовать не с "виртуальным" аддоном, а продемонстрировав взрослый пруф-оф-концепт.

karasevIA, 2021-11-29:
Я **не** имею отношения к команде Флиппера, а потому могу в более простой форме ответить на ваш комментарий, поясню почему я считаю что вы не правы в своем отношении к хакатону.

# почему standalone хранение - не актуально

Вы уже сделали (планируете сделать) решение на bluepill - вы молодец уже на этом этапе. Уже в этот момент проект полностью готов и функционален. У Ф103й есть USB и все функции будут выполняться на ура. Вы собираетесь подключаться к ПС по USB, а это означает что в вашем проекте даже аккумулятор бесполезен. Для вашей идеи идеально подойдет та самая bluepill, а флиппер гиперизбыточен. И, как правильно заметили выше, решение внутренними мощностями флиппера было бы хотя-бы более-менее интересно, на уровне со змейкой, шашками или чатом. 

# про мест нет - камень мимо огорода организаторов

В изначальном посте не было сказано, и в реальности критерием не была дата подачи заявки. Вам отказали потому-что идея не зашла и выше я описал более подробно возможные причины. Отбор проектов вынужденная мера и организаторы пошли на эту меру совсем недавно, когда количество людей перевалило за количество флипперов. Многие люди уже начали делать проекты и тратить на них свое время, и раннее предупреждение о том что их проект проиграл сэкономит это время всем участникам процесса. Будь ваша идея действительно стоящей, организаторы нашли бы ресурсы и на вас. Да и резерв - не самых худший из вариантов. Вот только идея - обьективно не стоящая. И даже то что вы трудились в поисках доступных чипов безопасности не делает ее стоящей.

# причем здесь я

Я просто комментатор, и у меня вызывает неприязнь клевета и выворачивание фактов. Вы можете делать это непреднамеренно, и серьезного ущерба в этом нет. Ну или не было бы, не будь у вас канала на более чем 700 подписчиков, и на который вы можете транслировать то же самое мнение, оклеветав команду организаторов хакатона. А там уже никто разбираться не будет кто прав а кто нет

iDoka, 2021-11-29:
@karasevIA если не имеете отношения к команде Флиппера, то зачем позволять себе вольность делать заявления от её лица?! Всё-таки в команде вы или нет?

----

> Для вашей идеи идеально подойдет та самая bluepill, а флиппер гиперизбыточен.

Флиппер, имея на борту BLE & NFC *идеален* для реализации холодного кошелька, увы в ограниченные сроки хакатона и учитывая текущее состояние API сделать интеграцию с мобилкой не представляется возможным.

----

Вообще откуда столько ненависти да еще и бездоказательное *обвинение в клевете*?!.. Такое ощущение, что это скрин переписки с вами я выложил и теперь вы мне мстите. А я всего-то лишь хотел привлечь внимание к тому факту - что некоторые члены комьюнити (с апрувленными идеями) собственно помимо идеи ничем более и не обладают (но тем не менее имеют "приоритет заявки") и теперь вот "попрошайничают" в личке участников группы "флиппер хакатон" - как то этот факт вы стыдливо не прокомментировали, типа всё ок?!..

karasevIA, 2021-11-29:
## заявления от её лица

Я **не делал заявлений** от лица команды флиппера, и **не имею отношения** к ней. Я такой же участник хакатона как и вы, а в данном треде я просто комментатор со своим мнением.

## сделать интеграцию с мобилкой

Вы сами на все ответили. Делайте свой кошелек на флиппере вне хакатона да и все, об это уже писали и это только приветствуется

## стыдливо не прокомментировали

   Фраза: "В изначальном посте ничего не было сказано что критерием для отбора проекта будет приоритет даты подачи заявки, но сейчас это выглядит так 🤷‍♂️ " - попытка клеветы. 
   Вы могли иметь ввиду одно (... но тем не менее имеют "приоритет заявки" ...), а в процессе изложения ваша мысль не идеально отобразилась, смешалась с небольшой каплей ненависти к тому что идея не зашла, и, итоговый результат получился каким получился. "Попрошайки" не заслуживают реакции, ни вашей ни моей, "смелое" комментирование ни к чему не приведет. И даже в вашем случае мне бы стоило пройти мимо если бы не одно но:
   Я увидел скрин вашего поста в чате (https://t.me/FlipperHackathon2021) , отношение к "попрошайкам" и желание "восстанавливать справедливость". Сложилось мнение, что когда вашу заявку полностью закроют, вы можете на всю свою аудиторию протранслировать то же мнение что и в вашем комментарии. Это, в отличие от балабольства "попрошаек", уже может быть серьезным ущербом репутации Флиппера. Повторюсь, что отношения к ним я не имею, но ребята трудятся, это видно по чату, видно по ишьюзам, видно по гитхабу проекта, и не они заслуживают даже вашего "но сейчас это выглядит так 🤷‍♂️", я уже не говорю про возможный пост у вас в канале (лишь мое предположение).

iDoka, 2021-11-29:
> В изначальном посте ничего не было сказано что критерием для отбора проекта будет приоритет даты подачи заявки
факт

>  но сейчас это выглядит так
моё персональное мнение

malmstrem, 2021-11-29:
Какая-то очень странная дискуссия выходит. 

> Вам отказали потому-что идея не зашла

Issue не закрыта, статус всё еще "на рассмотрении". Более того - если вы, @karasevIA, не один из членов команды флиппера, то откуда такие категоричные выводы, да еще и с указанием на то, что "идея не зашла"?

> Будь ваша идея действительно стоящей, организаторы нашли бы ресурсы и на вас

Опять же, вы беретесь судить о решениях организаторов, когда их мнение на этот счет неизвестно (по крайней мере в этом треде не было больше никаких комментариев). 

> Вот только идея - обьективно не стоящая

И снова то же самое. Для кого-то стоящая, для кого-то - нет. Решающим в конкретном случае является мнение организаторов, а если вы к ним не относитесь, то и рубить с плеча, что называется, рановато. 
По поводу клеветы и всего остального - Иван, вам не кажется, что вы несколько торопитесь с выводами? Есть канал в телеграме @iDoka, в который он выложил скрин с "диалогом" с одним из "попрошаек", окей, но что с того? Об этом никому нельзя было рассказывать? 

Не хочу никого обидеть, но в разрезе хакатона все эти комментарии выглядят малость неуместно - мне кажется, что транслировать тут своё мнение о том, что "идея не стоящая", да еще и основывая этот вывод на некоем "отказе", которого еще даже не было, несколько неуместно. Я ожидал, что дискуссии в комментариях будут идти о технической части проектов, о сложностях реализации, но уж никак не о каких-то глубоко субъективных опасениях о "репутационном ущербе" от сугубо гипотетических постов в личных телеграм-каналах (sic!).

karasevIA, 2021-11-29:
## категоричные выводы

В самом начале самого первого комментария я указал **"поясню почему я считаю что вы не правы в своем отношении к хакатону"**. И я добавляю исключительно свои выводы. Ни одного вывода или слова от имени команды флиппера я не делал, и 10 раз об этом писал, что уже чересчур. Ваш комментарий говорит о том что нужно было 20 раз писать о том, **что я никак не связан с командой флиппера**.

## их мнение на этот счет неизвестно

Организаторы скованы в своих высказываниях, утрируя, не могут отвечать всем подряд: "Извините, ваш проект говно". Даже быстро забраковать проект они не могут. Тем более у iDoka есть возможность изменить свою идею на реализацию без аппаратной части что, в теории, даст большую вероятность на прохождение в хакатон (на уровне со змейкой или шашками) и это, опять же, мое мнение, никак не связанное с командой организаторов, и основанное на поверхностных данных и первом комментарии.
Про телеграм канал я говорю только в контексте **возможного** поста. Ну а про остальное могу сказать лишь что тема была закрыта, комментариев не по техническим вопросам никто мы не писали. Вопреки вашему мнению в комментарии, вы:
- задаете нетехнические вопросы, вызывающие **нетехническую дискуссию**
- транслируете тут свое мнение **не по техническим** вопросам
- своим комментарием заводите дискуссию **"о каких-то глубоко субъективных опасениях"**

(0 о)

iDoka, 2021-11-29:
> в вашем случае мне бы стоило пройти мимо если бы не одно но: Я увидел скрин вашего поста в чате

Т.е. Efim Ilin (@sht4nigga) вовсе не в вашей команде  (проект  "Ethernet ping-тестер #29" ) и что именно он запостил скрин - **конечно же чистой воды совпадение**. Простите, но после такого лицемерия больше нет доверия ни к одному вашему "невинному" намерению.

Ефим, Иван, карма - вещь такая: никогда не знаешь когда в обратку прилетит.
Пожалуйста, сосредоточьтесь на своём проекте, начните уже созидать, нежели промышлять мелкими пакостями, прикрываясь при этом мнимой заботой о команде Флиппера.

karasevIA, 2021-11-29:
Здесь нет подвоха
Ефим в моей команде, и о вашем проекте я узнал из его сообщения в чате, в личных сообщениях вам  покажу что могу

И в чем тут лицемерие? Подтвердите хоть как то свои слова

aprosvetova, 2021-12-02:
40 из 41 участников одобренных команд зарегистрировались на мероприятие и получили билеты.
К сожалению, это даже больше, чем максимальная вместимость помещения, поэтому никого дополнительно взять не можем.

Не расстраивайтесь и не забрасывайте свою идею, пожалуйста. Мы рассчитываем, что Хакатон пройдёт хорошо, и тогда ещё не раз вернёмся к этому формату.

closed 2021-12-02 by aprosvetova