flipperdevices/flipper-hackathon-moscow #50: DOOM

br4wl0wl, 2021-11-27:
### Подробно опишите идею вашего проекта

Проект будет реализовывать DOOM-like шутер по технологии RAY-cast.

### Готовый проект будет уметь...

Будет один уровень с врагами, с возможностью ходить и стрелять, с отрисовкой на родном дисплее флиппера и управлением через родные кнопки.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@lavaFrai

M010, 2021-11-29:
Если не получится доделать на хакатоне -- готов помочь) Сейчас уже есть команда)

aprosvetova, 2021-12-02:
40 из 41 участников одобренных команд зарегистрировались на мероприятие и получили билеты.
К сожалению, это даже больше, чем максимальная вместимость помещения, поэтому никого дополнительно взять не можем.

Не расстраивайтесь и не забрасывайте свою идею, пожалуйста. Мы рассчитываем, что Хакатон пройдёт хорошо, и тогда ещё не раз вернёмся к этому формату.

closed 2021-12-02 by aprosvetova