flipperdevices/flipper-hackathon-moscow #51: Удаленный терминал RS-232

bondarevky, 2021-11-27:
### Подробно опишите идею вашего проекта

Хотел бы написать плагин для проброса Serial порта через Bluetooth + сделать аппаратный конвертер UART <-> RS-232.

### Готовый проект будет уметь...

Возможность подключения к флипперу по Bluetooth и передачи текстовых команд устройству с интерфейсом RS-232.
Настройка параметров передачи осуществляется на самом флиппере

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

iDoka, 2021-11-27:
Сама фича классная и я бы хотел такой пользоваться, вот только на момент, когда я последний раз интересовался существованием профиля напоминающего UART под BLE - такового не нашлось - каждый вендор свой велосипед изобретал (в отличие от serial port profile в Classic Bluetooth).
Если за это время ситуация изменилась - очень здорово.

aprosvetova, 2021-11-28:
@iDoka не изменилась, к сожалению. У нас в прошивке есть RPC, который работает в том числе по BLE через кастомную характеристику с костыльным самодельным flow control ＞﹏＜

aprosvetova, 2021-12-02:
40 из 41 участников одобренных команд зарегистрировались на мероприятие и получили билеты.
К сожалению, это даже больше, чем максимальная вместимость помещения, поэтому никого дополнительно взять не можем.

Не расстраивайтесь и не забрасывайте свою идею, пожалуйста. Мы рассчитываем, что Хакатон пройдёт хорошо, и тогда ещё не раз вернёмся к этому формату.

closed 2021-12-02 by aprosvetova