flipperdevices/flipper-hackathon-moscow #54: Приложение для 2FA

otopba, 2021-11-30:
### Подробно опишите идею вашего проекта

Мы хотим сделать приложение, в котором можно будет посмотреть 2FA ключ к одному из своих интернет аккаунтов.

Все мы пользуемся двухэтапной аутентификацией(я надеюсь). Кто-то использует Google Authenticator, кто-то Яндекс.Ключ, кто-то другие решения. Почти все они работают на [Time-based One-time Password Algorithm](https://ru.wikipedia.org/wiki/Time-based_One-time_Password_Algorithm) (TOTP). Используя наше приложение, можно будет авторизоваться на всех популярных интернет ресурсах. Все данные должны быть надежно зашифрованы. Хотелось бы попасть в основную прошивку

### Готовый проект будет уметь...

Нужно реализовать следующие фичи:

1) Добавление / удаление нового ресурса для входа по 2FA
2) Генерация OTP для выбранного ресурса
3) Блокирование входа в приложение паролем

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@emil151997

ojab, 2021-11-30:
Чому не U2F?

aprosvetova, 2021-11-30:
@ojab U2F мы штатно делаем

ojab, 2021-11-30:
Oh, nice!

aprosvetova, 2021-12-02:
40 из 41 участников одобренных команд зарегистрировались на мероприятие и получили билеты.
К сожалению, это даже больше, чем максимальная вместимость помещения, поэтому никого дополнительно взять не можем.

Не расстраивайтесь и не забрасывайте свою идею, пожалуйста. Мы рассчитываем, что Хакатон пройдёт хорошо, и тогда ещё не раз вернёмся к этому формату.

closed 2021-12-02 by aprosvetova