flipperdevices/flipper-hackathon-moscow #55: BLE remote

MindlessJoker, 2021-12-01:
### Подробно опишите идею вашего проекта

Эмуляция BLE клавиатуры флиппером для использования в качестве
- Media remote control
- PowerPoint presentation mode control

### Готовый проект будет уметь...

Флиппер сможет подключаться в качестве BLE HID устройства к хосту.
Из меню можно будет выбрать и использовать несколько типов клавиатур:
- для управления медиаплеером с клавишами управления громкостью/вперед/назад/пауза
- для управления режимом презентации: следущий слайд/предыдущий слайд/в начало/в конец
- стрелки + Enter/escape

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

### GitHub второго участника

@usoltsev

aprosvetova, 2021-12-02:
40 из 41 участников одобренных команд зарегистрировались на мероприятие и получили билеты.
К сожалению, это даже больше, чем максимальная вместимость помещения, поэтому никого дополнительно взять не можем.

Не расстраивайтесь и не забрасывайте свою идею, пожалуйста. Мы рассчитываем, что Хакатон пройдёт хорошо, и тогда ещё не раз вернёмся к этому формату.

closed 2021-12-02 by aprosvetova

usoltsev, 2021-12-02:
Ок, понятно.
В выходные встретиься можем - к нам или к вам, посмотрим на кухню, может
чего подержать могу))).
В субботу до 15 точно занят, в остальном пока жестких ограничений нету.
Маша от меня какого-то экшена по ремонту хотела, но пока конкретики не
знаю.(ёлка+подклейка мелочи + отвезти хлам в гараж)

MindlessJoker, 2021-12-02:
Хорошо, будем ждать следующей возможности. Успехов в проведении хакатона!