flipperdevices/flipper-hackathon-moscow #57: Использование Flipper в качестве носителя Рутокен

MisteRun, 2021-12-01:
### Подробно опишите идею вашего проекта

Хочу сделать возможным, использование Flipper'ом в качестве носителя Рутокен

### Готовый проект будет уметь...

В готовом проекте Flipper должен уметь работать как носитель Рутокен с возможностью выбора типа носителя (2.0, Lite, S и т.д.) В приоритете добиться работы с типом 2.0. Иметь удобный интерфейс менеджера выбора носителя и сертификата пользователя. Так же flipper должен иметь расширенный редактор носителя и сертификата (смена номера носителя + изменение пин-кода сертификата). Возможность создания виртуального носителя под нужными параметрами пользователя.

### Вы капитан команды?

- [X] Yes, I am the leader of my team

### Вы готовы приехать на Хакатон в Москву?

- [X] Yes, my team will attend

aprosvetova, 2021-12-01:
К сожалению, лимит на количество одобренных проектов превышен. Резерв тоже переполнен.

closed 2021-12-01 by aprosvetova

alexey-gamov, 2021-12-02:
Идея классная, надеюсь в будущем идею не забросите и реализуете!

Sobieg, 2021-12-03:
Классная идея!

MisteRun, 2021-12-03:
Жалко что не пригласили на хакатрон. Хотел детальнее описать проект и может найти команду для реализации этой идеи.

aprosvetova, 2021-12-03:
Нет мест.

Sobieg, 2021-12-03:
Я думаю, можно попробовать реализовать после получения устройств.

MisteRun, 2021-12-03:
Вот именно, приходится ждать. Другие же по блату получают устройства ранее и могут реализовывать свои и чужие ранее других))

aprosvetova, 2021-12-03:
Не по блату, а вовремя подав заявку.

Sobieg, 2021-12-03:
> Вот именно, приходится ждать. Другие же по блату получают устройства ранее и могут реализовывать свои и чужие ранее других))

Я думаю, идей на всех хватит. Кроме того, кажется, что целиком рутокен за несколько дней не сделать. Такие проекты могут и дождаться.